opendata-stuttgart/meta #12: Paper über Vergleichbarkeit von 15$ PPD42NS mit "professioneller" Hardware

mfa, 2015-04-04:
http://www.sciencedirect.com/science/article/pii/S0269749115000160

Kann das jemand bewerten?
Wie aussagekräftig ist das für Stuttgart?

mfa, 2015-04-04:
weiterer Artikel über Vergleichbarkeit mit dem ppd42ns: http://atmos-meas-tech-discuss.net/7/605/2014/amtd-7-605-2014.pdf

tomatitito, 2015-04-26:
**Zum Paper von Gao et al. (2015): A distributed network of low-cost continuous reading sensors to
measure spatiotemporal variations of PM2.5 in Xi'an, China**

In dem Paper kommen die PPD42NS ziemlich gut weg. Es werden 8 Sensoren getestet und mit verschiedenen teuereren Messgeraeten verglichen. Als Referenzgeraet wird ebenfalls ein optisches Instrument verwendet (TSI Dust Trak II Model 8532). Die Messwerte der PPD42NS und des Dust Trak sind vergleichbar, allerdings nicht einfach so "out of the box". Stattdessen muessen sie kalibiert werden, die Entwicklung eines *predictive models* nimmt einigen Platz in dem Paper in Anspruch. Die Beziehung der PPD42NS mit dem DustTrak ist nicht linear, deshalb wird ein polynomisches Regressionsmodell verwendet, um die Messwerte des Dust Trak durch die der PPD42NS "vorherzusagen". Obwohl die Messungen der acht einzelnen PPD42NS einen starken linearen Zusammenhang aufweisen (die Korrelation ist mindestens r=0.97), wird fuer jeden PPD42NS ein eigenes Vorhersagemodell spezifiziert.  

Die Studie wurde in Xi'an, China durchgefuehrt. Die Messwerte fuer PM2,5 in der Studie schwanken zwischen 85 und 140 mg/m^3, die Werte in Stuttgart sind niedriger. Das ist vor allem fuer die Kalibirierung wichtig, da die Sensoren nicht bei allen PM2,5-Konzentrationen gleich gut messen. Je hoeher die PM2,5-Konzentration ist, desto besser (weniger Fehlerbehaftet) sind die Messungen (Die Studie von Holstien z.B. findet lineare Zusammenhaenge mit dem verwendeten Referenzmessgeraet). Man kann die Ergebnisse deshlab nicht einfach uebernehmen. Tatsaechlich weisen die Autoren sogar daraufhin, dass die Sensoren in anderen Gebieten auf jeden Fall neu kalibriert werden sollten. Trotzdem ist die Studie auch fuer Stuttgart nicht nutzlos. Vor allem wird klar, dass die Kalibrierung und die Entwicklung eines predicitve models wichtig sind.

tomatitito, 2015-04-26:
Die Zusammensetzung von Feinstaub kann einen Unterschied fuer die Messung machen. Hier ein Artikel dazu:

[Cheng et al.(2015): PM2.5 and PM10-2.5 chemical composition and source apportionment near a Hong Kong roadway](https://www.dropbox.com/s/u0nkdvpyczyw8o9/Cheng%20et%20al_2015_PM2.pdf?dl=0)

tomatitito, 2015-04-27:
## Zum Artikel von [Holstius et al. (2015): Field calibrations of a low-cost aerosol sensor at a regulatory monitoring site in California](http://atmos-meas-tech-discuss.net/7/605/2014/amtd-7-605-2014.pdf)

### Was wird in der Studie gemacht?
In dieser Studie werden drei PPD42NS in Oakland, Kalifornien getestet und die Messwerte werden mit denen von verschiedenen anderen Instrumenten, darunter auch die offiziellen Werte des Bay Area Air Quality District (Referenzgeraet ist ein BAM 1020 von MetOne Instruments) verglichen. Um die Staerke des Zusammenhangs zu bestimmen werden Korrelationen und RSME (root mean square error of approximation) einmal auf Stundenbasis und einmal auf Tagesbasis (24h) bestimmt.

### Ergebnisse bezueglich der Vergleichbarkeit
Die Korrelation der drei PPD42NS mit dem BAM 1020 liegt zwischen r=0.55 - 0.60, die Korrelation auf Tagesbasis ist mit r=0.72 hoeher. Das ist aehnlich zu der [Studie von Gao et al. (2015)](http://www.sciencedirect.com/science/article/pii/S0269749115000160), in der auch staerkere Zusammenhaenge mit zunehmender Aggregation festgestellt wurden.

Der Zusammenhang mit dem Dust Trak Messgeraet, dass im [Paper von Gao et al. (2015)](http://www.sciencedirect.com/science/article/pii/S0269749115000160) als Referenzmessgeraet benutzt wurde, wird auch in diesem Artikel berichtet. Dieser Zusammenhang ist etwas groeßer als der mit dem BAM 1020. Es kommen auch zwei Messgeraete der Firma GRIMM zum Einsatz. Das ist insofern interessant, als das Landesgesundheitsamt Stuttgart auch Geraete dieser Firma benutzt. Die Uebereinstimmung der PPD42NS mit den GRIMM-Geraeten ist hoeher als die anderen, Korrelationen zwischen r=0.90-0.94.

### Inwiefern ist das wichtig fuer Stuttgart?
Die Konzentration von PM2.5in Oakland ist der in Stuttgart aehnlicher als die in Xi'an, China. Deswegen ist diese Studie fuer Stuttgart vielleicht etwas interessanter. Auch hier wird wieder klar, dass die Sensoren kalibriert werden muessen. Dazu sind *co-located measurements* notwendig. Man muss also die PPD42NS "for a sufficient length of time" (S.1126) neben ein anderes, offiziell verwendetes Messgeraet stellen. Wie lange das sein muss, haengt von der gewuenschten Qualitaet der Kalibrierung ab, die wiederum vom Zweck der Messungen abhaengt. Aber auch bei geringer Kalibrierungsphase koennte der PPD42NS zu Screening-Zwecken eingesetzt werden, um z.B. starke Veraenderungen der PM2.5-Konzentration festzustellen.

tomatitito, 2015-04-27:
Ausfuehrlicher Bericht ueber die unterschiedlichen Methoden zur Messung von PM2.5. Beinhaltet auch einige Vergleiche der verschiedenen Verfahren (Kap. 4):

[Guidance for using continous monitors in PM2.5 monitoring networks](http://www.epa.gov/ttn/amtic/files/ambient/pm25/r-98-012.pdf)

tomatitito, 2015-04-28:
Vergleich des GP2Y1010AU0F mit dem TEOM (Instrument zum messen der Partikelmasse):

[The Outdoor Dist Information Node - ODIN](http://files.figshare.com/1584339/Olivares_etal_ANZAW_2014_odin.pdf)

Fazit:
Nach Kalibrierung und unter Beruecksichtigung von Temperatur und relativer Luftfeuchtigkeit konnte Vergleichbarkeit hergestellt werden. Rohdaten des ODIN (Messgeraet, dass den GP2Y1010AU0F verwendet) waren nicht vergleichbar.

mfa, 2015-05-23:
http://indiaairquality.com/

tomatitito, 2015-05-25:
Auch interessant. Den Korrelationskoeffizienten beim Vergleich des 42FR mit dem Dylos finde ich allerdings nicht so aufschlussreich, da der Zusammenhang nicht linear zu sein scheint.