opendata-stuttgart/meta #14: Weiterer Vergleich von billigen mit professionellen Sensoren

mfa, 2015-04-04:
http://aqicn.org/sensor/shinyei/

mfa, 2015-05-03:
*Samyoung dsm501a* liegt vor und ist als dritter Sensor nun im Prototyp verbaut.

mfa, 2015-05-03:
Die beiden scheinen nicht vergleichbar zu sein.
Aus dem Artikel: *From the previous graph (Shinyei vs. Samyoung), it is clear that the data generated by the Samyoung sensor is not at all reflecting actual air quality.*

Marwe, 2015-09-17:
[Science article]
Environmental science: Pollution patrol
Step aside, fitness trackers. The next wave of personal sensors is giving people the ability to monitor the air they breathe.

DustDuino
http://www.nature.com/news/environmental-science-pollution-patrol-1.16654#sensors