opendata-stuttgart/meta #25: Feinstaubwerte in Zürich sind keine Tagesmittel. Wie werden sie ermittelt?

mfa, 2015-06-03:
* http://data.stadt-zuerich.ch/content/portal/de/index/ogd/daten/luftqualitaet_ugz_messungen.html#description1
* https://www.stadt-zuerich.ch/gud/de/index/umwelt/luft/messwerte/verlauf_24_stunden.html

tomatitito, 2015-06-05:
Hab bei der Stadt Zürich nachgefragt. Die haben ein OpenData Team(!), das sehr hilfsbereit war und meine Mail sofort an den entsprechenden Ansprechpartner weitergeleitet hat. Der hat mir dann auch sofort folgenden Link geschickt:

https://www.stadt-zuerich.ch/content/dam/stzh/prd/Deutsch/Statistik/Publikationsdatenbank/jahrbuch/2014/pdf/JB_2014_gesamt.pdf

Auf Seite 176 kann man alles nachlesen. Für PM10-Messungen kommen zwei unterschiedliche Verfahren zum Einsatz:
- Betastrahlen-Absorption (kalibriert auf das Referenzverfahren), misst Halbstundenmittel
- High Volume Sampler (das Referenzverfahren), misst 24h-Mittelwerte

PM2.5 wird wohl nicht gemessen. Was mir nicht klar ist, ist wie man die Halbstundenmittel anhand der 24h-Mittelwerte kalibrieren kann. Man darf aber bei Fragen nochmal nachhaken, das werd ich mal tun.

tomatitito, 2015-10-27:
Ich habe jetzt nochmal in Zuerich nachgefragt, wie genau die Kalibrierung gemacht wird. Der zustaendige Mann dort ist wirklcih sehr hilfsbereit und hat innerhalb kuerzester Zeit geantwortet.

Die Kalibrierung der Halbstundenmittel (gemessen mit Betastrahlen-Absorption) wird rueckwirkend fuer das letzte Kalenderjahr vorgenommen. Aus den Halbstundenwerten werden Tagesmittel berechnet. Diese Tagesmittelwerte und die 24h-Messungen des High Volume Samplers gehen dann in eine lineare Regressionsanalyse ein. Die resultierende Regressionsgleichung wird dann benutzt, um die Halbstundenwerte zu kalibrieren. Will heißen: die Halbstundenwerte werden in die Gleichung eingesetzt und das Ergebnis ist der kalibrierte Wert.

Das wird, wie schon gesagt, immer rueckwirkend fuer das Kalenderjahr gemacht. Fuer das laufende Jahr werden die Messwerte provisorisch mit der Gleichung des Vorjahres kalibriert.

Es scheint außerdem so zu sein, dass fuer jedes kontinuierliche Messgeraet (Betastrahlen-Absorption oder auch andere, es gibt also wohl noch mehr Messgeraete) ein eigener High Volume Sampler zur Kalibrierung verwendet wird.