opendata-stuttgart/meta #27: Arduino based box

Marwe, 2015-06-14:
http://joerg-krause.de/index.html

> Ich habe ein portables Gerät entwickelt dass die Konzentration von
> verschiedenen Gasen und von Feinstaub misst. Das Gerät hat Sensoren 
> für:
> 
> -Gase: CO, CH4(Gruppe) Methan, Propan, Butan usw. 
> -Alcohole: Formaldehyd, Toluene, Benzene, Alcohol, Aceton 
> (alles giftiges Zeug)
>  -Feinstaub > 1?m Partikelgröße (zum Vergleich Pflanzenpollen haben
> 10?m) -Temperatur und Luftfeuchtigkeit
> 
> Für alle Messgrößen sind Alarm Schwellwerte hinterlegt. Das Gerät ist
> sehr empfindlich. Wenn ich in der Nähe einen Wattebausch mit Spiritus
> ablege dauert es nicht lange und es geht ein Alarm los. Auch wenn man
> in der Nähe ein Teelicht anzündet steigt sofort der Feinstaubwert an.
> Das Gerät zählt die Rußpartikel der Kerze.

mfa, 2015-06-20:
Mit welchem Verfahren wird hier Feinstaub gemessen?

Marwe, 2015-06-26:
Im Video ist ein optischer Sensor üblicher Bauform zu sehen, aber welcher Sensor das genau ist, ist nicht beschrieben (evtl via PM anfragen).