opendata-stuttgart/meta #33: Nächtliche Peaks

Marwe, 2015-10-14:
Habe hier in den Daten nächtliche Peaks, das kann dann wohl nur Heizung sein, oder? Balkon ist straßenabgewandt. Komischerweise aber auch keine Verkehrs-Peaks tagsüber, obwohl signifikante Änderung des Verkehrs.

[image]

etho50, 2016-02-24:
Hallo Martin,

hat sich Deine Frage inzwischen erledigt? Wenn nicht: Die Peaks könnte eventuell an der damaligen Wetterlage gelegen haben. Ich habe den Ballonaufstieg von Stuttgart in dieser Zeit angeguckt. Nachts gab es Inversionen mit geringer Durchmischung.Tagsüber war die Durchmisschung aber immer gut.Es war eine ziemlich seltene Wetterlage mit Strömung aus dem Osten. Allerdings wären die ganz hohen Peaks damit nicht erklärbar. Vielleicht war es nur ein unterstützender Faktor. 
Mein Angebot steht: Wenn es um Meteorologie oder um Verarbeitung/Darstellung von Messdaten geht bin ich gerne behilflich. Ich habe dazu aus der beruflichen Vergangenheit viel Erfahrung. Komme einfach auf mich zu. Demnächst werde ich auch die Hardware für eine Station bei mir bestellen. 
Ich mache übrigens auch Vorhersagen der Schadstoffbelastung auf der Internet-Seite unserer Bürgerinitiative in Leonberg: www.agvl-leonberg.de

Grüße
Ewald 
p.s. Ich war gestern beim Treffen dabei und nutze diesen Kommentar auch als Test, nachdem ich mich bei Github angemeldet habe.

Marwe, 2016-03-03:
Hallo etho50,

erst einmal funkitoniert das mit den Issues ;)
Dann habe ich aktuell mal ein paar Plots der Datensätze vom archive erstellt, die ich hier hochgeladen habe:

http://mqtt.opendataset.click/output_plots/

Sind trotz etwas Filterung noch recht roh, und können noch verbessert werden, aber ein erster Wurf zur Visualisierung (mit R).

Marwe, 2016-03-03:
Und wegen der Wetterdaten würde ich gerne auch mal eine Hacking-Session machen, vielleicht kann man einigermaßen standardisierte Plots erzeugen und sich die Messungen mal ansehen und auf Plausibilität prüfen. 

Leider sind die Wettermessungen nicht im archive, da muss man wohl noch mal mit der django-API rumfummeln.