opendata-stuttgart/meta #43: Big vaccum cleaner to filter fine dust

Marwe, 2016-10-27:
http://www.france24.com/en/20161025-dutch-unveil-giant-vacuum-clean-outside-air
No comment

etho50, 2016-10-30:
klingt toll, aber hat 2 gewaltige Haken:

1. Der Energieverbrauch des Teils dürfte sehr hoch sein (--> klimaschädlich, Schadstoffausstoss bei der Stromerzeugung)
2. 30.000 m3 ist gerade mal eine Würfel von etwas mehr als 31x31x31 Metern! 
Frage: Wie viele Autos kann man in diesen Würfel stellen, damit die Maschine saubere Luft raus bläst? Wie sieht's mit dem Ultrafeinstaub aus? Wieviele Maschinen bräuchten wir, um die Luft einer Stadt zu reinigen? 

Fazit: Der Aufwand steht in keinem Verhältnis zum Ertrag. Es gibt keine Wundermaschinen. Das ganze Geo-Engineering nutzt nichts. Emissionen Einsparen ist die mit Abstand billigste und schnellste Lösung. es gibt dazu keine Alternative.

Marwe, 2016-10-30:
Das sehe ich auch so, ist wohl eher eine Anekdote der Technikentwicklung. Vielleicht ergeben sich aber in bestimmten, abgegrenzten Bereichen Einsatzmöglichkeiten.

> Wie sieht's mit dem Ultrafeinstaub aus?
>> "filtering out 100 percent of fine particles and 95 percent percent of ultra-fine particles"

Geoengineering ist generell schwierig und teuer (wer zahlt solche Technik?), daher bisher auch noch nicht wirklich in der Umsetzung.

ricki-z, 2016-10-30:
Für die ganze Stadt ist das wahrscheinlich nicht machbar. Aber warum nicht solche Geräte an den Tunnelausgängen anbringen. Dort ist die Luftbelastung meist deutlich höher als im Rest der Stadt. Der dort herausgefilterte Feinstaub kann dann schon nicht mehr weitergetragen werden. Am Neckartor würde sich das wahrscheinlich deutlicher bemerkbar machen als die komische Mooswand.

etho50, 2016-10-30:
1. Wieviele solcher Maschinen (die wahrscheinlich auch einen Höllenlärm machen) bräuchtest Du denn? Und was würde das kosten?
2. Für Tunnel gibt es viel effizientere Lösungen, Die Luft im Tunnel wird ja abgesaugt und kann in einem Abluftkamin ganz bequem wie in einem Kraftwerk gefiltert werden. Und  die Stickoxide können zu Gips verarbeitet werden. Das Problem in der Praxis sind nur die Kosten, weil der Staat so billig wie möglich baut und die Folgekosten keinerlei Rolle spielen. Aber diese Maschine bzw. die vielen Maschinen, die man bräuchte, sind bestimmt noch viel teurer (wahrscheinlich besonders in der Wartung).
Was die Mooswand betrifft, gebe ich Dir vollkommen recht.

ricki-z, 2016-10-30:
Gab inzwischen eine Kickstarter-Kampagne, um einen Prototypen in Rotterdam zu installieren. Dieser Prototyp ist freistehend und enthält 3 der "Staubsauger", Kosten 50.000$. Wenn man der Abbildung glaubt, sollte man sich daneben aufhalten können.
Kickstarter: https://www.kickstarter.com/projects/1777606920/the-smog-free-tower
Artikel mit Bild des fertigen Towers im Art Distrikt in Rotterdam: https://www.rt.com/news/364127-dutch-vacuum-air-filter/