opendata-stuttgart/meta #45: Auswertungen und Webseiten

Marwe, 2016-11-19:
Es gibt ein paar neue Visualisierungen und Ansätze für die Präsentation:

Interaktive Grafiken der Sensormesswerte von Fritz Mielert:
http://10ter-mai.de/feinstaub/chart/

Hackathon München Ergebnisse  (Jan Georg Plavec, Christian Frommeld et al):
http://community.globaleditorsnetwork.org/content/measuring-air-quality-germanys-dirtiest-town-0
https://drive.google.com/file/d/0BzBK1jhbvit0REdNM21sU0xwUFU/view

Marwe, 2016-11-19:
Signalqualität Übersicht:
https://www.madavi.de/sensor/signal.php

Marwe, 2016-11-22:
Und natürlich die Leonberger Initiative AGVL:
http://ethoma.de/AGVL_Home/index.php/luftreinhaltung/

polygontwist, 2017-02-04:
http://10ter-mai.de/ scheint kaputt zu sein

mielert, 2017-02-05:
Hey, 
http://10ter-mai.de/ muss fritzmielert.de heißen (die Domain 10ter-mai.de habe ich schon über ein Jahr nicht mehr). Allerdings bin ich gerade mitten im Umzug zu feinstaub-stuttgart.info (siehe auch https://github.com/mielert/feinstaub-dataviz).
Herzliche Grüße
Fritz

Marwe, 2017-03-28:
http://feinstaub.rexfue.de/
Vis für zwei Sensoren