opendata-stuttgart/meta #47: Mailingliste einrichten

Marwe, 2016-12-01:
Eine Mailingliste für Neuigkeiten und Austausch zwischen den Beteiligten (Paten, Entwickler, ...) einrichten. Kann eventuell organisatorisch über den shack mailman gemacht werden, wenn dort ein entsprechender Antrag gestellt wird.

mielert, 2016-12-05:
Sollen wir nicht versuchen, bei der wunderbaren Domain luftdaten.info zu bleiben? Da wird doch garantiert auch ein Mailman laufen.

Marwe, 2016-12-13:
Über luftdaten läuft soweit ich weiß nur die Webseite und nichts sonst. Ein mailman nur für diese Mailingliste wäre sicher overhead...

mielert, 2016-12-13:
Also, wenn Ihr Serverplatz inkl. Mailman braucht. Bei mir läuft das standardmäßig.

ricki-z, 2016-12-14:
Der Server, auf dem luftdaten.info läuft, sollte das können. Jan? David?

noobilicious, 2016-12-14:
Es müsste überprüft werden, ob eine Python Umgebung auf dem Server läuft. Dafür müsste Jan schauen, welchen Vertrag er beim Hoster hat.

fuergestalten, 2016-12-14:
Python ist dabei.

Ryuno-Ki, 2016-12-14:
Welche Version von Python?

Marwe, 2017-11-27:
Mail ist wohl eher out, closing this

dhebbeker, 2017-11-27:
Ich finde die Idee eine Mailingliste einzurichten immer noch gut. Denn E-Mail ist wahrscheinlich der kleinste gemeinsame Nenner der Kommunikationskanäle alle Paten und Entwickler.

pathmapper, 2017-11-29:
Hier ist noch ein issue zum Thema offen: https://github.com/opendata-stuttgart/luftdaten.info/issues/5

marians, 2018-02-09:
Was ist denn der aktuelle Stand bezügl. Community-Kommunikationskanal?

Ich würde einen solchen Anlaufpunkt sehr begrüßen. Die Wachsende Anzahl von Menschen, die zum Beispiel einfach nur eine Station betreiben wollen und dabei vor weitgehend wiederkehrenden Fragen/Problemen steht, sollte einen Anlaufpunkt haben. Es kann nicht sein, dass man sich erst mal über Kontakte durchfragen muss, wer denn was wissen könnte. Ein [Verweis auf die OKLabs](https://twitter.com/Jan_sagt/status/961665050154012674) greift da auch ein bisschen kurz.

Wie in Issue https://github.com/opendata-stuttgart/luftdaten.info/issues/5 geschrieben, gibt es Slack. Aber da ist mir als Besucher völlig unklar, wie ich Zugang erhalte. Wenn das der Kommunikationskanal für die Community sein soll, müsste mindestens irgendwo (auf luftdaten.info) erklärt werden, wie man da rein kommt.

Im Sinne der Skalierung dieses Projekts appelliere ich dringend an die vielbeschäftigten, dem Thema mehr Priorität zu geben. Sonst werdet Ihr früher oder später zum limitierenden Faktor für den Erfolg dieses Projekts, in das Ihr doch so viel Zeit und Liebe steckt.

Nordrunner01, 2018-02-10:
Moin
Sehr schöner input. Ich habe mich auch öfter gefragt wie man wohl mal eben auf die Schnelle kontakt aufnehmen kann. Wie wäre es mit einer Matrixinstanz? Dezentral, nicht kommerziell und open source. Unsere Gemeinschaft betreibt auch einen Matrixserver, welcher auch über ein Webfrontend zu erreichen ist : https://matrix.to/#/#iotde:matrix.ffslfl.net Ein Kanal wo es um IoT und solche Dinge geht. U.A. auch den FSS von denen wir auch welche betreiben. Gruß

pathmapper, 2018-02-10:
> Im Sinne der Skalierung dieses Projekts appelliere ich dringend an die vielbeschäftigten, dem Thema mehr Priorität zu geben. Sonst werdet Ihr früher oder später zum limitierenden Faktor für den Erfolg dieses Projekts, in das Ihr doch so viel Zeit und Liebe steckt.

Schön auf den Punkt gebracht @marians .

Der Wunsch nach einem Kommunikationskanal für die Community wurde schon öfters geäußert, z.B. auch hier

- https://github.com/opendata-stuttgart/meta/issues/78
- https://github.com/opendata-stuttgart/luftdaten.info/issues/5
- https://github.com/opendata-stuttgart/sensors-software/issues/170#issuecomment-346340401

@ricki-z gibt es aktuell irgendwelche Überlegungen/Aktivitäten in diese Richtung?

ricki-z, 2018-02-10:
Wir suchen gerade eine Foren-Software, die relativ sicher und einfach wartbar ist. Hosting sollte bei uns (oder zumindest in Deutschland) sein, damit wir dank neuer Datenschutzgrundverordnung nicht gleich zum Abmahnziel werden. Womit viele Cloudlösungen rausfallen.
Zu Mailinglisten: Die Admins sollten alle Mails (eines oder mehrerer Channels) bekommen, um das ganze unter Kontrolle zu haben (siehe letzter Absatz). Was für uns dann keine Änderung beim Mailaufkommen bedeutet. Eher das Gegenteil.
Zu Matrix: Ich habe gerade auf die verlinkte Seite geschaut. Da bekomme ich erstmal gesagt, ich brauche eine App. Da wären dann die meisten Nutzer schon raus.
Zu Reddit: Es gibt wohl Leute, die damit zurecht kommen. Ich sehe eine Reddit-Seite und bin weg. Ich kann da die Struktur dahinter nicht finden ...
Zu Wikis: Wenn nur bestimmte Personen editieren können, können wir bei den FAQ auf Luftdate.info bleiben. Wenn wir es freigeben, muss auch hier wieder kontrolliert werden (siehe letzter Absatz).

Wir haben Nutzer von 9 bis 90 Jahren, die mit dem Kommunikationskanal klarkommen müssen.

Zum Thema Kontrolle: Wir bekommen schon jetzt zum Teil so unverschämte Mails, das alle 'offenen' Kommunikationsformen auf jeden Fall kontrolliert werden müssen. Es gibt Personen, die sich für die totalen Cracks halten und dann an irgendeinem Schritt scheitern und uns dann als totale Deppen bezeichnen. Es gibt die Leute, die von einem 30-Euro-Sensor Messwerte erwarten, die mindestens so genau sind wie die eines 30.000-Euro-Geräts und dann nicht verstehen, warum sie mit den Messwerten jetzt nicht SOFORT vor Gericht gehen können. Und so weiter ...

Nordrunner01, 2018-02-10:
Moin. 
Wir bieten allen Menschen und Gemeinschaften die Möglichkeit. Webseite, Forum und Wiki sind vorhanden. 
Schaut doch mal rein : iot-usergroup.de .

Gruß

ricki-z, 2018-02-10:
Hallo @Nordrunner01,

schau mal in dein Postfach. Dort schreibe ich dir, warum ich deinen Kommentar editiert habe.

Gruß Rajko

Adorfer, 2018-02-14:
Falls ihr jemanden sucht, der Übung mit discourse hat (das ist das, was für https://forum.freifunk.net hüte) und Euch bei der Einrichtung hilft (oder wenn ihr eine VM dafür gespendet braucht von unserem lokalen Freifunk-Verein (hier: neanderfunk.de), dann sagt Bescheid. Wir haben Übung in der Wartung von Dutzenden von VMs inkl. deren Monitoring. 
Moderation würde ich aber gern jemandem "vom Kernteam" überlassen, sprich: Hillfestellung bezieht sich nur auf Setup, Maintenance und ggf. VM-Resourcenspende. 

Vorteil an Discourse ist, dass "Mailingslisten-Affine" Nutzende das Ding auch "nur per Mail" benutzen können. d.h. man kann Unterforen mit einer dedizierten Mailadresse kompeln, Mail-Notification und Mailreply ist problemlos. Selbst neue Threads sind per Mail anlegbar.

marians, 2018-02-14:
+1 für Discourse. Es ist der würdige Nachfolger der allzu verbreiteten *bboard Foren, die trotz unsäglicher Usability immense Verbreitung erfahren haben.

Alternative wäre IMHO die gute alte Mailingliste, am einfachsten über Google Groups.

Adorfer, 2018-02-21:
Alternativ können wir prüfen, ob wir in dem bestehenden Freifunk-Forum einen Unterbereich "Luftdaten" (oder meinetwegen "opensensors", Vorschläge willkommen) einrichten. 
Dann ist keine (evtl. zu vermeidende) direkte "mamentliche Verknüpfung" mit "luftdaten.info" gegeben. Und (bezahlte) TrollInnen würden das nicht so als lohnende Beute betrachten. Oder würden zumindest auf ein tiefenentspanntes Moderatorenteam treffen. 
Nur eigentlich wäre es schöner, wenn -früher oder später- das doch als "was eigenes" laufen würde.