opendata-stuttgart/meta #49: Muss das WLAN immer an sein ?

Theodin, 2017-01-21:
Ich spiele mit dem Gedanken, hier bei mir einen Feinstaub-Sensor an den Balkon zu hängen und habe dazu 2 Fragen:
Unser Wlan ist nachts meistens aus. Loggt das Gerät die Werte weiter (denk ich mal) und schickt diese dann, wennd as Wlan wieder da ist, an den Server?
Wir haben zwar draußen eine Steckdose, bei der weiß ich aber nicht, wie das mit dem Dauerbetrieb insbesondere bei Regen ist. Wie ist das bei bestehenden Sensoren gelöst mit der Stromversorgung?

Ansonsten komm ich mal ins Shackspace und frage euch da.

tokudan, 2017-01-21:
Ich habe auch nur eine "Schönwetter"-Steckdose auf dem Balkon, daher kommt USB für mich auch nicht in Frage.
Vorschlag beim Congress war ein flaches USB-Kabel zu benutzen das durch ein geschlossenes Fenster geht, aber das klappt bei mir leider auch nicht wegen der Balkontür.
Mit 6 AA Akkus in Reihenschaltung (je 2A) hält der Sensor leider grade mal ca. 18h durch, der braucht also schon recht viel Strom. Ich versuche trotzdem grade meinen Sensor auf Akkubetrieb umzurüsten.

ricki-z, 2017-01-21:
Wir bevorzugen zwar Sensoren, die 24 Stunden laufen, da auf der NodeMCU kaum Platz ist, um etwas zu speichern. Aber wenn du mit Lücken in den Daten leben kannst, sollte es kein Problem sein, den Sensor trotzdem aufzunehmen.
Was den Akku-Betrieb angeht: es gibt wohl Leute, die an einer gepufferten Solar-Lösung arbeiten, aber bisher habe ich aus der Richtung leider kein Feedback bekommen.

ffrow, 2017-01-26:
Wie sieht es mit der Möglichkeit aus, den Sensor mit einer Powerbank zu betreiben und den Upload über ein UMTS WLAN Modem (z.B.) Huawei E5832 zu machen (ebenfalls Powerbank unterstützt). Damit sollte man dann auch eine mobile Lösung hinbekommen, oder?

Wenn man an Hauptverkehrsstraßen messen will, also z.B. an Fahrradwegen neben der Fahrbahn, dann müsste man dort einen Tagesverlauf von Beginn Berufsverkehr bis Ende Berufsverkehr aufzeichnen.

Die Powerbanks gibt dann wiederum mit integrierter Solar-Nachladung. So sollte man dann die Zeiten ausdehnen können.

Hört sich das an, wie etwas, das funktionieren könnte?

Marwe, 2017-01-27:
* Powerbanks schalten meist aus, wenn nur geringer Strombedarf ist (<Ladeströme): ich fand bislang nur eine, die durchgehend läuft.
* Solarnachladung: gibt es da etwas funktionierendes?
* Modem kann man nicht direkt verbinden, müsste also unabhängig eine Verbindung/WLAN bereitstellen, so wie der genannte Router
* eventuell will man noch die Position des Sensors messen/Updates pushen (GPS Daten, müssten in der DB für den Sensor freigeschaltet sein), ricki-z hat eine Smartphone-App angefangen, mit Smartphone könnte man auch GPS und Uplink erledigen

polygontwist, 2017-02-04:
solar funktioniert bedingt, ich habe ein 75W-Modul (recht groß, Raum Rostock, Südausrichtung)  im Winter kommt aber wenig Strom an - reicht gerade für 5-7h.

ElFlosen, 2017-02-06:
Idee zur Lösung des Speicherplatzes auf der NodeMCU:
Könnte man nicht einen speicher-shield mit SD-Karte anbringen und bei fehlendem W-Lan die Daten dort zwischenspeichern? Dann könnte nach dem Wiederverbinden der Upload der fehlenden Daten initiiert werden.

Marwe, 2017-02-07:
SD Karte hilft deshalb nicht für die API, weil der Zeitstempel beim einliefern gesetzt wird.
Grundsätzlich ist es aber möglich, eine SD-Karte zu betreiben, man muss allerdings etwas suchen, bis man für den ESP funktionerenden Code findet, es gibt da ein Fork-Desaster ;)
Aber für den eigenen Gebrauch soweit machbar, dann muss man übrigens entsprechend die Firmware patchen und die entsprechenden PINs (um)belegen.

georg90, 2017-07-17:
Hi!
Ich komme zum nächsten Workshop in Hamburg und wundere mich doch stark darüber, dass der Sensor im Dauerbetrieb läuft. 
In den Einstellungen ist der Intervall auf 145 Sekunden eingestellt, wenn ich dem Guide hier folge: https://openhomeautomation.net/esp8266-battery/ kann durch den deep sleep des esp8266 modules einiges an Energie gespart werden und der Sensor mit einer relativ kleinen Batterie/Akku mehrere Monate betrieben werden.

Gibt es die Möglichkeit den Sensor mit Batterie/Akku zu betreiben? Wie hoch ist der Gesamtstromverbrauch?

ricki-z, 2017-07-17:
Es gibt Probleme mit dem Deep Sleep und dem SDS011. Der SDS011 startet wieder, wenn die NodeMCU in den Deep Sleep geht. Dadurch würden wir dann ca. 50 mA für die laufende NodeMCU gegen 100 mA für den laufenden SDS011 tauschen. Das macht wohl auch wenig Sinn.
Der Deep Sleep Modus ist des weiteren wie ein Reset. Dadurch müssten wir die Programmstruktur anders aufbauen. Und wir haben einige Nutzer, die z.B. für Smart Home Lösungen auf die Daten des Sensors direkt zugreifen wollen. Aktuell liegt die Anzahl dieser Benutzer deutlich höher als die Anzahl derer, die unbedingt eine Solar-Lösung haben möchten.

georg90, 2017-07-18:
Danke für die schnelle Rückmeldung. Dachte mir schon, es liegt an dem Sensor. Dann bekommt er halt ein Kabel :)

pathmapper, 2017-09-19:
@Marwe 

> Powerbanks schalten meist aus, wenn nur geringer Strombedarf ist (<Ladeströme): ich fand bislang nur eine, die durchgehend läuft.

Könntest du vielleicht mitteilen, welche Powerbank durchgehend läuft? Meine derzeitige schaltet auch ab.

Marwe, 2017-09-20:
Ich habe wohl dieses Modell hier:
https://www.aliexpress.com/item/20000mAh-External-Solar-Power-Battery-Charger-For-iPhone-Samsung-Cell-Phone-Tablet/1000001331677.html

Die genannten Werte sind natürlich Quatsch (mAh), und mit dem Solarpanel lässt sich auch nicht wirklich genug laden.

pathmapper, 2017-09-21:
Danke! 
Kannte ich bisher gar nicht, diese Powerbank-Solarmodul-Kombination.
Wenn das Quatsch ist suche ich eine reine Powerbank ohne Solarmodul die nicht abschaltet, am liebsten mit Lithium-Ionen-Akku.

dokape, 2017-09-23:
Die Diskussion hat sich vom Topic (Wlan 24h an?) wegbewegt.
zusammengefasst: 
- Der Sensor puffert aufgrund der geringen Speicherkapazität keine Daten.
- Wer 24/7 Daten benötigt, sollte das WLAN 24h verfügbar haben.
- Die Luftdaten.info Seite zeigt nur Sensordaten an, die in den letzten 5 Minuten geliefert wurden
- Mit einem mobilen Access-Point (z.b. Smartphone-Hotspot) können auch mobil Daten erfasst und übertragen werden.
- zu empfehlen wäre bei mobilen Einsatz auch die Verwendung eines GPS wie dem unterstützten NEO6M.

Die Diskussion zu Solar und Powerpacks erfolgt hier:
https://github.com/opendata-stuttgart/sensors-software/issues/99

Kann das issue hier geschlossen werden? 
Die Fragen zum Topic sollten ausreichend beantwortet sein.