opendata-stuttgart/meta #50: Weitere Messwerte BMP180?

networxnet, 2017-01-24:
An einer Stelle wird erwähnt das Auslesen eines BMP180 wäre im Code implementiert, es wird jedoch nirgens weiter darauf eingegangen.
Gibt es hierzu weitere Infos?

Marwe, 2017-01-24:
Siehe
https://github.com/opendata-stuttgart/sensors-software/blob/master/esp8266-arduino/ppd42ns-wificonfig-ppd-sds-dht/ext_def.h

```
// BMP180, Luftdruck-Sensor
#define BMP_READ 0
#define BMP_API_PIN 3
#if defined(ESP8266)
#define BMP_PIN_SCL D4
#define BMP_PIN_SDA D3
#endif
```
SCL, SDA ist also i2c.

Der müsste dann noch von uns in der Datenbank angemeldet werden, bzw. wird auf madavi direkt visualisiert, z.B.:
https://www.madavi.de/sensor/graph.php?sensor=esp8266-13601904-bmp

Ansonsten ist der einzige, der meines Wissens diese Sensoren hat: ricki-z

ricki-z, 2017-01-24:
Die Beschreibung oben ist soweit richtig. Zu verwenden ist die I2C Version des BMP180. Die Anschlüsse sind wie oben geschrieben SCL an D4 und SDA an D3.
Die benötigten Biliotheken und Funktionen sind in der Firmware dann schon drin, es muss lediglich in der WiFiConfig der Eintrag BMP180 aktiviert werden. Dann werden die Daten an die ausgewählten APIs übertragen. Bei der api.luftdaten.info muss uns das nur mitgeteilt werden, da die Daten nur bei einem entsprechenden Sensor-Eintrag in der Datenbank entgegen genommen werden.

Ansonsten: software issues bitte in das entsprechende Repo sensors-software

Und wenn soweit alles klar ist, Issue bitte schließen.

networxnet, 2017-01-26:
Gibt es weitere Info zu der verwendeten Bibliothek? Wird nur der BMP180 unterstützt?

networxnet, 2017-01-26:
Schon gefunden, ist nur die Adafruit BMP085 Lib.

ricki-z, 2017-01-26:
Arbeite gerade an der Integration des BME280, damit könnte der DHT22 und der BMP180 ersetzt werden.

networxnet, 2017-01-26:
SEHR GUT! Genau das war auch mein Plan.

Der BME280 scheint ein Top-Sensor zu sein und ich denke die Metrologischen Daten Temp/Feuchte/Luftdruck ergeben im Zusammenhang mit der Feinstaubmessung viel Sinn.

ricki-z, 2017-01-27:
Die aktuelle Version im Repo unterstützt jetzt den BME280. Wenn die Tests erfolgreich sind, geht diese Version am Wochenende auf die Sensoren.

ricki-z, 2017-02-07:
BME280 (I2C) Version ist auf den meisten Sensoren angekommen.

closed 2017-02-07 by ricki-z