opendata-stuttgart/meta #51: Alternative Feinstaubsensoren? Z.B. Sharp

networxnet, 2017-01-26:
Hat jemand weitere alternative Sensoren getestet? 
Z.B. den Sharp GP2Y1010AU0F, preislich unter 15€ bei einer Lebensdauer von 5 Jahren und bereits ab 2,5V lauffähig.
Hört sich für mich jetzt erstmal vorteilhaft an...

ricki-z, 2017-01-26:
Wir haben zuerst mit einem ähnlichen Sensor (PPD42NS) experimentiert. Alles ohne Ventilator hat eigentlich einen viel zu ungleichmäßigen Luftzug, um irgendwie vernünfig zu messen.
Der Sharp baut auf den Luftzug der Klimaanlage, den man dann "nachbauen" müsste. Macht zusätzliche Teile, Kosten und Fehlerquellen.
Die richtige Technik fürs Feinstaubmessen zu erklären, braucht etwas Zeit bzw Platz. Schreibe mich vielleicht nochmal per Mail an.

networxnet, 2017-01-27:
Gern oder wie schauts mit einem Chat aus?

Marwe, 2017-01-28:
Von den PPD habe ich auch noch ein paar herumliegen :/

irunonayran, 2017-04-11:
@ricki-z: könnt ihr das Gesprächsprotokoll oder die Mail veröffentlichen? Hab mich auch schon gefragt welche Vor- und Nachteile der sds011 hat und womit man die kalibriert und wie teuer bessere sind.

dokape, 2017-09-23:
Inzwischen wird neben dem SDS011 und dem PPD42NS auch die PMS x003 Familie unterstützt. 
https://github.com/opendata-stuttgart/sensors-software/issues/103
Die Diskussion über die Software erfolgt dort:
https://github.com/opendata-stuttgart/sensors-software/issues

Hier wäre vorrangig die Diskussion zur Dokumentation und Beschreibung.