opendata-stuttgart/meta #52: NodeMCU ESP8266 5v vs 3.3v Betrieb

thaxy, 2017-01-30:
Die “Original-Firmware” schaltet nach Auslieferung auf den Pins 5V, was zur Beschädigung des SDS011 führt.

Verstehe ich richtig, dass die customized Firmware den Ausgang der Pins trotz 5V input auf 3.3V drosselt?
Oder sollte man der ganze NodeMCU möglichst nur mit 3.3V versorgt werden?

Danke für die Hilfe.

Marwe, 2017-01-30:
Das sind wahrscheinlich keine 5V, sondern 3V3, es muss sich dabei um die Pins für Tx Rx handeln, da diese die einzigen sind, an denen sich der Output ändern kann. VU und GND 

Das esp Modul selbst muss immer mit 3V3 laufen, auf dem Nodemcu-Board gibt es einen 5V->3V3 Konverter, daher sind die 5V USB OK, der ESP darauf läuft so mit 3V3.

ricki-z, 2017-01-30:
Die NodeMCU Boards sind meines Wissens nach alle auf 5V ausgelegt und müssen deshalb auch so betrieben werden. Ein Spannungswandler setzt das dann für den ESP8266 auf 3.3V.
Der Grund für den Hinweis war, das wir einige SDS011 zerstört haben beim Versuch diese an ungeflashten NodeMCUs zu betreiben. Die Vermutung war, das an einem der Pins für den SDS011 in der ausgelieferten eine Spannung anliegt, die eine der Schutzdioden am SDS011 zerstört.
Auch wenn wir inzwischen davon ausgehen, das wir mit unserer Vermutung nicht richtig lagen, wollen wir einfach sicher gehen, das die NodeMCUs einfach beim Start einen uns bekannten Zustand haben. Denn was auf den NodeMCUs bei Auslieferung läuft, kann sich jederzeit ändern. Und damit auch die Zustände der Pins (Input/Output/PWM ...).

polygontwist, 2017-02-04:
hallo, ich hatte den Sensor dran ehe ich den Hinweis auf der Webseite gelesen hatte (könnte prominenter sein). ich hatte auch rx und tx vertauscht - der Sensor hat es überlebt.

closed 2017-02-07 by ricki-z