opendata-stuttgart/meta #57: Infos zu eigenen Sensoren

ElFlosen, 2017-02-06:
Noch ein Verbesserungsvorschlag für http://luftdaten.info/feinstaubsensor-bauen/
(gehört der Vorschlag hier hin?)

Habe vor einigen Tagen meinen Sensor fertig zusammengebaut. Mir fehlte aber im Anschluss an die Anleitung was ich nun mit den Daten selber anfangen kann bzw. wie ich meine Daten visualisieren kann.

- Durch Suche im Internet habe ich herausgefunden, dass ich auf https://www.madavi.de/sensor/graph.php?showfloat meinen Sensor suchen und mir dort die Daten als chart anzeigen lassen kann.
Sollte man ggf. auf der Homepage erwähnen bzw. falls andere Optionen vorhanden sind diese auch aufführen.

- Ich glaube in einer älteren Anleitung habe ich gesehen, dass es bei der Konfiguration des Feinstaubsensors auch die Möglichkeit gab per USB die Daten abzuspeichern. In meiner Version konnte ich das nicht mehr auswählen. Wurde das entfernt?

- Ich kenne mich leider nur sehr wenig mit Programmierung aus, habe aber verstanden, dass man die Daten auch an eine eigene API-Schnittstelle übertragen kann. Danach habe ich festgestellt, dass man seine Dropbox ebenfalls "API-befähigen" kann. Gibt es eine Anleitung wie man die Daten der Sensoren auf die eigene Dropbox bekommt? Dies wäre eine super Möglichkeit um die Daten "der breiten Masse", die keine Programmierkenntnisse hat, zugänglich zu machen.

Gruß
Florian

polygontwist, 2017-02-06:
Hallo, 
zum Thema Programmierung, ich habe in der Konfig meinen eigenen Server eingetragen 
```
[x]Senden an eigene API
Server: [rp1] (ich hab einen Raspi, IP ist dynamisch)
Pfad:   [feindata.php]
Port:  [80]
```

und lasse die Daten von einem php-Script _(feindata.php)_ verarbeiten:
```
<?php 
$logURL="feinstaub.log";//aktueller Messwert
$logToOrdner="log/".date('Ymd').".log";//Messwert pro Tag in eine Datei schreiben, die im Ordner log liegt

$itime= time();//timestamp Januar 1 1970 00:00:00 GMT
$datum= date('Y-m-d');
$zeit=  date('G:i:s');

$daten = file_get_contents('php://input'); //hier kommen die Daten vom Sensor als JSON an

//aktuelles Messwert
$handle=fopen($logURL,'w');
fwrite ($handle, "{" );
fwrite ($handle, '"time":'.$itime.',');
fwrite ($handle, '"datum":"'.$datum.'",');
fwrite ($handle, '"zeit":"'.$zeit.'",');
fwrite ($handle, '"daten":'.$daten  );
fwrite ($handle, "}".chr(10) );
fclose ($handle);

//als Datensätze in Ordner, 
$add=file_exists($logToOrdner);
$handle=fopen($logToOrdner,'a');
if($add)fwrite ($handle, "," );
fwrite ($handle, "{" );
fwrite ($handle, '"time":'.$itime.',');
fwrite ($handle, '"datum":"'.$datum.'",');
fwrite ($handle, '"zeit":"'.$zeit.'",');
fwrite ($handle, '"daten":'.$daten  );
fwrite ($handle, "}".chr(10) );
fclose ($handle);
?>
```
Dann habe ich mir mit Javascript ein widget geschrieben der die Daten ausließt und anzeigt.

Der Datensatz vom Sensor sieht, je nach Sensoren, so aus (hab momentan nur den Luftgütesensor drann):
```
{
	"software_version": "NRZ-2016-048", 
	"sensordatavalues":[
		{"value_type":"SDS_P1","value":"17.30"},
		{"value_type":"SDS_P2","value":"14.10"},
		{"value_type":"samples","value":"327970"},
		{"value_type":"min_micro","value":"165"},
		{"value_type":"max_micro","value":"26494"},
		{"value_type":"signal","value":"-63 dBm"}
	]
}
```

Das kann man machen wenn man einen eigenen Server (z.B. Homepage, NAS oder Raspi) mit php hat.

dokape, 2017-09-17:
Auf 
http://deutschland.maps.luftdaten.info/#6/51.165/10.455
kannst du auf deinen Sensor reinzoomen. 
Die Angabe der geolocation in der URL ermöglicht sogar ein Abspeichern der Map inkl. Location und zoomlevel.

Auf http://feinstaub.rexfue.de/ kannst du deinen (oder auch andere Sensoren) angeben und bekommst entsprechend Charts 24h, 7 Tage, 30 Tage, 365 Tage.

Mit diesem Script auf einem Webserver (im WWW oder im lokalen Raspi) kannst du die Daten als csv-daten ebenfalls selbst speichern und für deine Bedürfniss auswerten.
https://github.com/opendata-stuttgart/madavi-api/blob/master/data_simple.php

Hast Du noch weitere Fragen dazu oder kann diese Diskussion geschlossen werden?

ElFlosen, 2017-09-21:
Danke für eure Antworten. Werde mich bei Zeiten mal mit rasberry pi und der entsprechenden Programmierung beschäftigten müssen. Diskussion wird geschlossen. Viele Grüße

closed 2017-09-21 by ElFlosen