opendata-stuttgart/meta #58: Mobiles Messgerät "Pierre Track"

Marwe, 2017-02-07:
Die Firma http://www.airney.de/ bringt ein tragbares Messgerät für Luftqualität auf den Markt, "Pierre Track" genannt, das neben Feinstaub PM1/PM2.5/PM10 auch andere Parameter bestimmt.

In einem Video ([Youtube channel](https://www.youtube.com/channel/UCWerPZJS6rjSVs_So3iWJlw)) wird die Messung folgender Parameter gezeigt:
* Temperatur [°C]
* relative Luftfeuchtigkeit [%]
* TVOC (volatile organic components) [ppb]
* CO2 [ppm]
* O2 [%]
* Luftdruck [mbar]
* PM 1,0 [µg/m³]
* PM 2,5 [µg/m³]
* PM 10 [µg/m³]

Marwe, 2017-02-07:
Es wird heute einen Beitrag im MDR zum Thema geben:
Einfach genial | 07.02.2017 | 19:50 Uhr Feinstaubmessung per Rucksack
http://www.mdr.de/einfach-genial/eg-moderation-232.html

dokape, 2017-09-17:
@Marwe 
Danke für deine Infos. 
Da ich keine weitere Diskussion hier sehe, kannst Du bitte das Issue schließen? 
Es geht nicht verloren und wird nur archiviert.

closed 2017-09-17 by Marwe