opendata-stuttgart/meta #60: Keine Daten

svenkubiak, 2017-03-06:
Ich bin gerade etwas ratlos. Seid einiger Zeit versuche ich nun schon einen NodeMCU mit SDS011 und DHT22 in Betrieb zu nehmen. Ich bin strickt nach Anleitung vorgegangen und habe erst die Firmware geflasht und dann die Sensoren angeschlossen - alles gem. Anleitungen. Die Web-Oberfläche kann ich zwar aufrufen, aber ich sehe keine Daten, weder für den SDS011, noch für den DHT22. Der SDS011 geht auch an (man hört den Lüfter). Wenn ich beide Sensoren über Beispiel-Programme anspreche (also ohne die Web-Oberfläche) bekomme ich Daten. Allerdings muss ich dafür die Sensore an andere Pins anschließen. Kann das Problem was mit einem falschen Pin-Mapping zu tun haben? Siehe dazu: https://github.com/esp8266/Arduino/issues/584

Bin für jeden Tipp dankbar!

ricki-z, 2017-03-06:
Hallo Sven,
hast die Daten deines WLANs schon eingetragen? So lang der Sensor kein WLAN hat, werden keine Daten erfasst.

svenkubiak, 2017-03-07:
Ja, ich hatte bisher immer eine direkte WLAN Verbindung zum NodeMCU aufgebaut. Aber auch mit Einbindung in das eigene WLAN, sehe ich keine Daten für den SDS011 pder dem DHT22.

[image]

ricki-z, 2017-03-07:
Kannst du mir einen Screenshot der Konfigseite schicken? rajko (at) codefor.de . In solchen Fällen wäre übrigens die ChipID sehr hilfreich.

svenkubiak, 2017-03-07:
Kommando zurück. Wahrscheinlich war ich nach dem ganzen Testen in den letzten Tagen zu ungeduldig. Nach ein paar Minuten werden Daten angezeigt! :-)

Kurze Anschlussfragen:
- Das 5GHz WLANs nicht unterstützt werden ist Hardware bedingt oder aktuell nicht implementiert?
- Gibt es ein Beispiel wie der Request aussieht der an die eigene API gesendet werden?

Vielen Dank und viele Grüße,
Sven

ricki-z, 2017-03-07:
5 GHz wird von der NodeMCU nicht unterstützt, das Fehlen ist also hardwarebedingt.
Den Request kann man sich ansehen, wenn man sich über die Serielle Konsole der Arduino IDE die Debug Ausgaben ansieht. Der Request ist ein POST an die angegebene URL und enthält einen JSON String, der ungefäht so aussieht (Beispiel mit SDS011 und DHT22):
{"software_version": "NRZ-2016-047", "sensordatavalues":[{"value_type":"SDS_P1","value":"66.04"},{"value_type":"SDS_P2","value":"53.32"},{"value_type":"temperature","value":"22.30"},{"value_type":"humidity","value":"34.70"},]}
Die Sensor-ID wird im HTTP-Header-Feld 'X-Sensor' übertragen.

svenkubiak, 2017-03-07:
Perfekt! Danke die Untersützung!

closed 2017-03-07 by svenkubiak