opendata-stuttgart/meta #61: HTTPS Verbindung für Custom API führt zu Reset

svenkubiak, 2017-03-09:
Ich würde die Daten meiner Sensor gerne auch an eine eigene API schicken. Die Übertragung via HTTP funktioniert auch ohne Probleme. Beim Senden via HTTPS quittiert mir der ESP8266 allerdings den Dienst mit folgendem Stack Trace und resettet sich.

```
## Sending to custom api: 
Start connecting to apps.svenkubiak.de

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000200 depc=0x00000000

ctx: cont 
sp: 3fff2b70 end: 3fff30e0 offset: 01a0

>>>stack>>>
3fff2d10:  00000080 3fff616c 4021ba30 4023cc87  
3fff2d20:  3fffb47c 3fff616c 3fff5d44 4023ce20  
3fff2d30:  00000000 ffffffff 3fffb46c 3fff616c  
3fff2d40:  00000080 3fff616c 3fff8c84 4023d21d  
3fff2d50:  3fffbe94 3fff616c 3fff8c84 4023dc15  
3fff2d60:  3fff61cc 15c5807a b1e12040 00000001  
3fff2d70:  3fff616c 3fff8c6c 3fff8c84 00000001  
3fff2d80:  00000010 3fff616c 0000000f 4023de89  
3fff2d90:  3fff5d44 3fff6714 0000000f 00000001  
3fff2da0:  00000001 3fff675c 3fff6a48 4023ccec  
3fff2db0:  00000200 3fff66cc 3fff68ef 00000000  
3fff2dc0:  00000200 3fff66cc 3fff68ef 4023ef15  
3fff2dd0:  3fff2e00 00000000 000001d0 00000030  
3fff2de0:  371acc6f a22244ba 36f08101 00000004  
3fff2df0:  00000004 3fff6614 3fff68e9 4023b020  
3fff2e00:  6c4f0203 584dd1c4 3a9fe810 18590fa3  
3fff2e10:  a825d6ff 7d3e18e9 9e2780ff e51440ce  
3fff2e20:  a817bed6 a7014028 cd3a611f cc6fb4c5  
3fff2e30:  00000004 3fff68e9 3fff6614 4023b421  
3fff2e40:  cc61a938 b6f2a9d3 5bd42b90 3fff5efc  
3fff2e50:  00000000 00000004 00000004 402118d3  
3fff2e60:  3fff68ed 3fff68e9 3fff83d4 00000c40  
3fff2e70:  00000cb2 3fff758c 0000000e 00000004  
3fff2e80:  3fff6614 3fff68e9 00000004 00000004  
3fff2e90:  00000004 3fff68e9 3fff6614 4023ae18  
3fff2ea0:  00000000 3fff68e9 3fff6614 4023b1c0  
3fff2eb0:  00000012 3ffe83d4 00000013 401004d8  
3fff2ec0:  3fff2f10 0000000e 00000010 00000000  
3fff2ed0:  3fff5ff4 3ffe83d4 3fff6614 01000000  
3fff2ee0:  3fff2f10 3fff2f60 3fff6614 4023af70  
3fff2ef0:  40211754 00002a78 3fff20c0 00001387  
3fff2f00:  0000ee9b 3fff2f60 3fff5e84 40211d45  
3fff2f10:  000001bb 3fff655c 3fff2f60 4021117e  
3fff2f20:  c640ee59 00000001 3fff2f88 3ffe8370  
3fff2f30:  000001bb 3fff2f60 3ffe83d4 40211f55  
3fff2f40:  3ffea2d8 c640ee59 3ffea2d8 c640ee59  
3fff2f50:  3ffe9839 3ffe83d4 3fff3088 4020f072  
3fff2f60:  3ffe9a90 00000000 000003e8 03a29fb9  
3fff2f70:  3fff13b0 3fff5efc 3fff5e84 00000000  
3fff2f80:  00000000 00000000 3fff6494 000000bf  
3fff2f90:  000000bc 000004d9 3fff2fe0 4021adf5  
3fff2fa0:  000001bb 00000000 3fff414c 4021b0c0  
3fff2fb0:  3ffe985d 3ffe934e 3fff2fe0 03a29fb9  
3fff2fc0:  00000000 3ffe934e 3fff2fe0 4020ff0d  
3fff2fd0:  3fff2fe0 00000030 00000010 ffffffff  
3fff2fe0:  3fff60e4 0000001f 00000010 00000000  
3fff2ff0:  00000000 00000000 00000000 00000000  
3fff3000:  00000000 3fff5dfc 0000000f 00000007  
3fff3010:  3fff5de4 0000000f 00000000 3fff5eb4  
3fff3020:  0000000f 00000000 3fff5e9c 0000000f  
3fff3030:  00000000 3fff5f8c 0000005f 00000057  
3fff3040:  3fff5f34 0000004f 0000004e 3fff5d2c  
3fff3050:  0000000f 00000000 3fff602c 000000af  
3fff3060:  000000aa 3fff4164 0000000f 00000000  
3fff3070:  3fff6254 0000009f 00000091 3fff596c  
3fff3080:  0000000f 00000000 3fff62fc 0000018f  
3fff3090:  00000189 00000000 feefeffe feefeffe  
3fff30a0:  feefeffe 3fff414c 00000000 000001bb  
3fff30b0:  00000000 00000000 00000001 3fff20b8  
3fff30c0:  3fffdad0 00000000 3fff20b0 4021ba7c  
3fff30d0:  feefeffe feefeffe 3fff20c0 40100718 
```

Die Daten sind die gleichen wie bei der HTTP-Verbindung, allerdings anstatt Port 80, entsprechend Port 443.

ricki-z, 2017-03-09:
Auf Grund des geringen RAMs der NodeMCU funktionieren nur Zertifikate bis 2048 Bit.
Bei 4096 Bit stürzt die NodeMCU ab.

svenkubiak, 2017-03-10:
Ein selbstsigniertes Zertifikat mit 2048 Bit würde also funktionieren?

ricki-z, 2017-03-10:
Das sollte eigentlich funktionieren. Sowohl Luftdaten.info als auch die Madavi API nutzen ein 2048 Bit Zertifikat.

svenkubiak, 2017-03-10:
Perfekt. Selbst-signiertes Zertifikat mit 2048 Bit funktioniert. Danke für die Unterstützung.

closed 2017-03-10 by svenkubiak