opendata-stuttgart/meta #62: Datensatzbeschreibung

gmaubach, 2017-04-07:
Hallo zusammen,

die TroLUG (Troisdorfer Linux User Group) diskutiert derzeit über den Bau Eurer Sensoren. Ich möchte mich dem Projekt aus Sicht der Datenanalyse anschließen. 

Frage 1:
In Euren Sensordatenverzeichnissen, habe ich keine Datensatzbeschreibung gefunden. Könnt Ihr einen Link darauf veröffentlichen?

Frage 2:
Der Blick in die Sensordatendateien zeigt eine teilweise uneinheitliche Datenablage. In einigen Dateien, finde ich den Spalte, die ich für die Spalte mit den Feinstaubmesswerten halte (siehe Frage 1) statt Dezimalwerten Datumswerte. Beides in einer Spalte scheint mir auf inkonsistente Datenablage hinzudeuten. Könnt Ihr schon Aussagen zur Qualität der gesammelten Daten treffen?

Ich freue mich über Eure Nachricht.

Viele Grüße

Georg

ricki-z, 2017-04-09:
Schnelle Antwort auf Frage 2: Wir speichern die CSV mit Punkt als Dezimaltrennung. Bei einem deutschen Excel führt dies dazu, das die Werte dann als Datum angezeigt werden, wenn sie im entsprechenden Bereich liegen. Das liegt also nicht an inkonsistenter Datenablage.
Bei Excel hilft es, die Datei in '.txt' umzubenennen und dann beim Öffnen (in Excel, nicht via Doppelklick) in Schritt 3 unter 'Weitere ...' das Dezimaltrennzeichen auf Punkt und den 1000er-Trenner auf Komma zu setzen.

dokape, 2017-09-23:
Die Daten sind auf 3 Wegen zugreifbar.
.csv-Daten können von der Projekthomepage heruntergeladen werden.
Der Sensor kann die Daten auf einen eigenen Server übertragen (.z.b. lokaler Raspberry pi, apache-server auf eigenem Webspace). Dazu steht z.b. bei der Madavi-Api ein PHP-Scripft simple-data zur verfügung.
Das speichert die im json Format gelieferten Daten als .csv-Datei lokal ab. Mit etwas programmiergeschick kann das auch in eine Datenbank, Monodb, mysql etc zur eigenen Auswertung gespeichert werden.
Als dritten kann der Sensor auch direkt abgefragt werden. 
http://"ip-adresse-sensor"/data.json
liefert die aktuellen Daten im json-Format zur eigenen freien Weiterverwendung.

Eine Dokumentation der Datensatzbeschreibung ist derzeit nicht näher verfügbar.(2do) 

Noch Fragen dazu? 
Kann dieses issue hier geschlossen werden? 

Die Sensorweiterentwicklung kann auch dort verfolgt werden.
https://github.com/opendata-stuttgart/sensors-software/issues