opendata-stuttgart/meta #63: Plastikschlauch

kratz00, 2017-04-11:
Ihr erwaehnt an zwei Stellen einen _20cm Plastikschlauch_ (1, 2).
Ich denke der Schlauch ist als Luftzufuhr fuer den Feinstaubsensor gedacht.
Beim Zusammenbau der Komponenten wird der Schlauch nirgends erwaehnt, auch auf den Bildern ist dieser nirgends zu sehen.

Braucht man den Schlauch oder nicht?

(1) https://github.com/opendata-stuttgart/meta/wiki
(2) https://github.com/opendata-stuttgart/meta/wiki/Bezugsquellen-Einzelteile

ricki-z, 2017-04-12:
Der Schlauch ist zu empfehlen. Damit wird in den 10 Sekunden "Warmlauf" sicherer Umgebungsluft angesaugt. Ohne Schlauch wird zunächst nur die Luft im Gehäuse angesaugt.
Auf der Seite luftdaten.info ist der Schlauch ganz oben im Bild zu sehen. Ich werde die entsprechende Stelle der Doku aber noch ergänzen.

kratz00, 2017-04-18:
Danke @ricki-z, ich habe mir heute im Baumarkt 1m Plastikschlauch gekauft (gibt es da nur als Meter-Ware). Auf luftdaten.info habe ich gar nicht geschaut gehabt, sondern nur hier im Wiki. Wie du schon selber festgestellt hast, auch auf luftdaten.info wird der Schlauch beim Zusammenbau der Komponenten nicht erwaehnt.

dokape, 2017-09-17:
@kratz00 
Sind noch Fragen bzgl. des Schlauchs oder kannst du die Diskussion schließen?

kratz00, 2017-09-17:
Von mir aus kann die Diskussion geschlossen werden.
Ich hatte sie noch als Erinnerung fuer @ricki-z offen gelassen:

> Ich werde die entsprechende Stelle der Doku aber noch ergänzen.

dokape, 2017-09-23:
Beschreibung

kratz00, 2017-09-23:
Super, vielen Dank @dokape !

closed 2017-09-23 by kratz00