opendata-stuttgart/meta #65: Falsche Zeitzone in den Messwerten

manne01, 2017-04-16:
Guten Abend,

wovon hängt denn die Zeitzone ab, die in den Messdaten mitgeliefert wird?
Bei mir ist es aktuell +2 Stunden Unterschied zu den Messwerten.

Kann ich das irgendwo einstellen?

Beste Grüße
Euer Manfred

tokudan, 2017-04-16:
Bei zwei Stunden Differenz wird vermutlich UTC (https://de.m.wikipedia.org/wiki/Koordinierte_Weltzeit) übermittelt. Damit wird die Zeitzone vermutlich auch nicht auf dem Sensor einstellbar sein, weil der Sensor die ganzen Zeitzonen nicht kennt. Sollte jedes System das deine Zeitzone unterstützt einfach umrechnen können, wenn du mehr Details lieferst kann ich vielleicht mehr sagen.
Notfalls: für Winterzeit eine Stunde dazu rechnen, für Sommerzeit zwei Stunden.

Disclaimer: bin selbst nur Nutzer des Sensors und habe am der Firmware nicht mitgearbeitet.

manne01, 2017-04-16:
Ich habe nochmal weiter geforscht und muss mich entschuldigen. Die Zeitzone bzw. Uhrzeit ist korrekt. Ich hatte in meinem Auswertungsscript einen fiesen Fehler drin der die Ursache war.

Vielen Dank für die Antwort und nochmals Entschuldigung!

closed 2017-04-16 by manne01

ricki-z, 2017-04-16:
To answer the question: we are saving time stamps in UTC. Timestamps are saved server side. The NodeMCU doesn't have a real time clock.