opendata-stuttgart/meta #67: OLED

blebbens, 2017-06-18:
Hi,

Nach längerer Zeit habe ich meinen Luftsensor wieder aktiviert.

Ich hatte damals ein schönes 3D-Gehäuse mit für Sensor und OLED gedruckt.
Mit der neueren Firmware scheint das OLED nicht mehr aktivierz werden zu können. Es zeigt nun meine Werte mehr an, sondern "no updates available".

Warum zeigt es die Messwerte nicht mehr an?

ricki-z, 2017-06-18:
Bitte warte ca. 3 Minuten bis die erste Messung durchgeführt wurde. Wir haben die Länge des Messzyklus auf 2,5 Minuten erhöht, um mehr Lebenszeit für den SDS011 zu erreichen.
Bis zur ersten Messung wird die letzte Meldung angezeigt. In deinem Fall also "no update available".

blebbens, 2017-06-18:
Okay... ein Modus, in dem das WLAN-Modul nicht genutzt wird, wäre wünschenswert - beispielsweise 23-6 Uhr, in der WLAN abgeschaltet ist.

ricki-z, 2017-06-18:
Bitte entweder Titel ändern oder neuen 'Issue' beginnen. Danke.

dokape, 2017-09-17:
@blebbens 
Besteht das Problem mit der Anzeige noch weiterhin oder wird nach 146 Sekunden auf dem Display die Werte angezeigt?
Wenn das Anzeigeproblem nicht mehr besteht, bitte diese Diskussion schließen.

closed 2018-07-08 by ricki-z