opendata-stuttgart/meta #68: SDS011 - Ventilator läuft nicht

zakkk99, 2017-07-02:
Hi,
habe das Problem dass der Ventilator des SDS011 nicht läuft. Wenn ich das USB Kabel anschließe zuckt der Ventilator ganz gut und bewegt sich dann nicht mehr. Die Feinstaubmessung wird vorgenommen. Macht aber wohl keinen Sinn wenn der Ventilator nicht läuft? Woran kann das liegen? Ventilator defekt?
Danke für eure Antwort!

P.S. SDS01 ist am NodeMCU über Vin und GND angeschlossen. Die Pins A0, GND, VU heissen bei meinem NodeMCU A0, RSV, RSV. Denke bei dieser Version ist Anschluss über Vin und GND richtig oder?

ricki-z, 2017-07-03:
Hallo,
der Ventilator ist mit einer Steckverbindung angeschlossen. Falls möglich könntest du also den Stecker abziehen und den Ventilartor einmal direkt mit 5 Volt versorgen. Wenn er dann nicht läuft, Ist wohl der Ventilator defekt.

dokape, 2017-09-17:
@zakkk99 Hattest Du Erfolg? Läuft es jetzt? 
Besteht noch Diskussionsbedarf oder kann diese Diskussion geschlossen werden?

closed 2018-02-18 by ricki-z