opendata-stuttgart/meta #69: Own API, get no Data

RockdenWok, 2017-07-10:
Try to receive Data on my Own Webserver.
Configured to my IP/data.php Port 80.

data.php:
`<?PHP
                $fp = fopen("/var/www/data.txt","a");
                fputs($fp,"-----" . date("d.m.y H:i",time()) . "---\n");
                foreach($_POST as $key => $value) {
                        $line = $key . ":" . $value . "\n";
                        fputs($fp,$line);
                }
                $retVal = json_decode($_POST);
                        fputs($fp,"--- TRY JSON ---\n");
                        fputs($pf,$retVal);
                fclose($fp);
?>
`
Result data.txt
`
root@server:/var/log/apache2# cat /var/www/data.txt
-----10.07.17 12:51---
--- TRY JSON ---
-----10.07.17 12:54---
--- TRY JSON ---
-----10.07.17 12:56---
--- TRY JSON ---
root@server:/var/log/apache2#
`
Apache.log 
`root@server:/var/log/apache2# cat access.log | grep 9.164
10.9.9.164 - - [10/Jul/2017:12:24:54 +0200] "POST /data.php HTTP/1.1" 200 153 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:27:21 +0200] "POST /data.php HTTP/1.1" 200 153 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:29:46 +0200] "POST /data.php HTTP/1.1" 200 153 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:32:12 +0200] "POST /data.php HTTP/1.1" 200 153 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:34:37 +0200] "POST /data.php HTTP/1.1" 200 153 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:37:03 +0200] "POST /data.php HTTP/1.1" 200 153 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:39:29 +0200] "POST /data.php HTTP/1.1" 200 153 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:41:54 +0200] "POST /data.php HTTP/1.1" 200 153 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:44:20 +0200] "POST /data.php HTTP/1.1" 200 150 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:46:45 +0200] "POST /data.php HTTP/1.1" 200 150 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:49:11 +0200] "POST /data.php HTTP/1.1" 200 150 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:51:36 +0200] "POST /data.php HTTP/1.1" 200 150 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:54:02 +0200] "POST /data.php HTTP/1.1" 200 150 "-" "-"
10.9.9.164 - - [10/Jul/2017:12:56:28 +0200] "POST /data.php HTTP/1.1" 200 150 "-" "-"
root@server:/var/log/apache2#
`

Whats wrong ?

ricki-z, 2017-07-10:
Your foreach statement walks through $_POST and not the json. So replacing $POST with $retVal should solve the problem. For a working example look at https://github.com/opendata-stuttgart/madavi-api/blob/master/data_simple.php

RockdenWok, 2017-07-10:
OK, thanks for Answer and the Example.
$_POST run trough an possible existing Array send by post.
This was necessary because i did not know you send URL Encoded or JSON.

fputs($fp,var_export($retVal));
sould run trough an possible json Array. But there are two failures into my code.
  $retVal = json_decode(file_get_contents("php://input"));
sould be :
  $retVal = json_decode(file_get_contents("php://input",**true**));

and :
  fputs($fp,var_export($retVal));
sould be
  fputs($fp,var_export($retVal),**true**);

Both missing true Flags set the Value to an PHP valid Array
Code that works and retuns the whole Array :

```
<?PHP
        $retVal = json_decode(file_get_contents("php://input"),true);
        $fp = fopen("/var/www/data.txt","a");
        fputs($fp,"-----" . date("d.m.y H:i",time()) . "---\n");
        fputs ($fp,$retVal["esp8266id"] . "\n");
        fputs ($fp,$retVal["software_version"] . "\n");
        foreach ($retVal["sensordatavalues"] as $sensordatavalues) {
                $values[$sensordatavalues["value_type"]] = $sensordatavalues["value"];
        }
        $line = var_export($values,true);
        fputs($fp,$line);
        fclose($fp);
?>

```

Now i see all sendet Values and Array Keys.
Perhaps it will be an good idea to link the example to the API Section of the Wiki.

dokape, 2017-09-24:
Section / Description API expanded. More infos provided. 

I think, this issue can be closed.

closed 2018-02-10 by ricki-z