opendata-stuttgart/meta #70: WLAN Verbindung verloren / Sensor reagiert nicht mehr?

geektoor, 2017-08-23:
Servus,

ich hab aktuell das Problem, das der Sensor bei "inaktivitaet" die WLAN Verbindung zu verlieren scheint, bzw. einfach "haengt". Der Sensor laeuft stabil solange man ihn z.B. per ICMP Ping beschaeftigt. Laesst man den Ping weg, dauert es nicht lange und der Sensor ist nicht mehr erreichbar (liefert scheinbar auch keine Daten mehr).
Ist das schon irgendwem so ergangen? Gibts da evtl. nen Tweak zu um das zu verbessern? Firmware Autoupdate ist aktiv und der Stand sollte daher aktuell sein.

Gruss + Danke,
Sven

geektoor, 2017-08-24:
Ahoi,

hab mal weiter versucht das Problem einzugrenzen. Es sieht so aus, als wuerde der Sensor bzw. NodeMCU nicht mehr auf Arp Anfragen antworten. Der Router bekommt die DNS Requests fuer die API Hosts, fragt dann aber zurueck "hey, welche MAC hat denn die IP?", darauf kommt aber keinerlei Antwort. Das erklaert auch grob, warum mit dem Ping der Sensor stabil funktioniert.
Ich hab jetzt noch nicht in den Code geschaut, aber vermutlich reagiert die Firmware nur unter bestimmten Umstaenden aufs Arp. Solange die MAC im Cache ist, ist alles super.
Ich geh mal weiter suchen :)

Gruss,
Sven

dokape, 2017-09-23:
Hallo Sven.

Schau mal im Bereich für die Sensor-Software.
https://github.com/opendata-stuttgart/sensors-software/issues

Hier ist vor allem die Diskussion über Beschreibung und Dokumentation. 

Hast Du Dein Problem inzwischen weiter eingrenzen oder gar lösen können? 
Wenn ja bitte Lösung posten und schließen.
Wenn das Problem noch weiter besteht, kannst Du im Bereich Sensor-Software 
dein Anliegen noch einmal beschreiben? 
Evtl. findest Du dort auch in den bereits geschlossenen Issues eine Hilfestellung.

dokape, 2017-09-26:
So einen Effekt hatte ich gestern plötzlich auch.
War aber wohl das erste Mal, dass das auftrat. Bisher war das bei mir stabil. 
https://github.com/opendata-stuttgart/sensors-software/issues/144
pingen ging noch. Alle anderen Anfragen abgelehnt bzw. keine Reaktion.