opendata-stuttgart/meta #73: Displays

pathmapper, 2017-09-17:
Laut FAQ http://luftdaten.info/faq/#toggle-id-11 können Displays mit dem Treiber-Chip SSD1306 angeschlossen werden.

Dazu folgende Fragen:

- Welche Display sind empfehlenswert? Gibt es neben dem Treiber-Chip SSD1306 weitere Mindestanforderungen (z.B. Anzahl der Pixel)?
- Was wird auf einem angeschlossenen Display standardmäßig angezeigt?
- Gibt es kompatible Displays, die ohne weitere Modifikation für den Außeneinsatz geeignet sind? Falls ja welche?
-  Gibt es eine maximale Kabellänge, damit ein angeschlossenes Display noch funktioniert?

dokape, 2017-09-17:
Ich habe beim Chineses folgendes Display bestellt

[image]

pathmapper, 2017-09-17:
Danke, hilft mir weiter!
Ja, dachte an direkten Außeneinsatz aber Gehäuse für das Display ist auch OK.

closed 2017-09-17 by pathmapper