opendata-stuttgart/meta #74: BMP280 und DHT22 zeigen jeweils 5 crad zuviel an

sohnkarlsruhe, 2017-09-22:
Hallo, habe mir einen Sensor gebaut aber die Sensoren zeigen ungenaue Werte was die Temperatur angeht an. Luftfeuchtigkeit stimmt einigermaßen. Habe jeweils den BMP und DHT ausprobiert. Was kann ich tun?

dokape, 2017-09-23:
Hallo,
willkommen beim Feinstaubprojekt :-)
Hier ist die Diskussion zur Dokumentation und Beschreibung des Projektes. 
Fragen zur Software werden hier diskutiert:
https://github.com/opendata-stuttgart/sensors-software/issues

Auch ich habe unterschiedliche Anzeigen bei DHT und in meinem Falle BME.

sohnkarlsruhe, 2017-09-25:
Der 3. Sensor zeigt jetzt die richtigen Werte an... man muss wohl 10 Stück kaufen und hoffen, dass ein paar funktionieren.

closed 2017-09-25 by sohnkarlsruhe