opendata-stuttgart/meta #78: Forum

sareiizi, 2018-01-10:
Beim Slack kann man anscheinend nur mit bestimmten eMailadressen teilnehmen? Ein öffentliches Forum wäre Klasse!

sareiizi, 2018-01-22:
Wie wäre es mit einem Reddit?

ricki-z, 2018-01-22:
Dann lieber kein Forum ;-)

sareiizi, 2018-01-31:
Vielleicht hast du ja einen besseren Vorschlag. Ich fände ein Reddit zum Thema Feinstaub schon sinnvoll.

Nordrunner01, 2018-02-10:
Moin.
Wie wäre es mit einem vorhandenen Forum? Eigene Rubrik? Noch relativ neu und alles im Fluß und möglich.
Schaut euch doch mal um. Zu finden unter : https://forum.iot-usergroup.de/
Ich war mal so frei euer klasse Projekt ein wenig zu beschreiben dort : https://forum.iot-usergroup.de/t/feinstaubsensor-mit-dem-esp8266/134

Gruß