opendata-stuttgart/meta #80: Mirroring archive.luftdaten.info using rsync

ignisf, 2018-02-02:
Hello,

Have you considered simplifying the mirroring of archive.luftdaten.info by adding for example rsync access to it?

ricki-z, 2018-02-07:
Hi,
it is best pratice to reduce the offered services to a minimum. Rsync access would open just another possibility for hackers.

closed 2018-02-07 by ricki-z

ignisf, 2018-02-07:
Hello @ricki-z,

1. rsync is basically a standard amongst linux distros for mirroring their repositories. Debian, Gentoo, Arch, OpenWrt to name a few... I consider it a tested an proven technology.

2. It is MUCH faster and more convenient than running `wget --mirror` against archive.luftdaten.info daily.

3. It will save both the clients and archive.luftdaten.info *much* bandwidth.

ricki-z, 2018-02-07:
Hi,
1. this is true, but for us it is just another service which we have to look at (updates, monitoring)
2. the difference in data volume shouldn't be that high if run daily. wget can use gzip compression and both have to copy the same new files. wget will send HEAD requests for all files. That may need some time and bandwidth. But for this you can use wget on yesterdays directory (wget --mirror -np http://archive.luftdaten.info/`/bin/date -d yesterday +%Y-%m-%d`/). Convenience: how many users use rsync regularly and don't know how to use wget ?
3. look at 2.