opendata-stuttgart/meta #81: Übergreifende Dokumentation der APIs, Archivdaten und unterschiedlicher IDs

ebner, 2018-02-07:
Mein Use Case ist folgender: ich möchte an die Sensordaten eines bestimmten Landes kommen, inkl. historischer Daten. Da ich über die APIs nur die letzten Messungen bekomme muss ich also auch auf das Archiv zugreifen. Dabei habe ich entdeckt dass API und Archiv unterschiedliche IDs verwenden. Die Seite https://github.com/opendata-stuttgart/meta/wiki/APIs hilft mir leider nicht weiter. Da Airtube.info Daten zu haben scheint welche ich auch benötige muss es also einen Weg geben um da irgendwie ran zu kommen.

Ich benötige folgendes:

- Alle Sensor-IDs mit einem bestimmten Länderkürzel. Ich kann natürlich http://api.luftdaten.info/static/v1/data.json fragen und alle Sensoren mit country: 'XY' herausfiltern, aber die IDs die ich bekomme stimmen nicht mit dem Archiv überein.
- Filter-Queries via http://api.luftdaten.info/v1/filter/ bei der ich auch nach Land (und nicht nur nach Lat/Lon) filtrieren kann.
- Eine Möglichkeit um die ID der Messstation (NodeMCU ID) zu den Sensor IDs (die auf maps.luftdaten.info verwendet werden) und den IDs der API zuordnen zu können. Ich habe keine Information darüber gefunden wie man von den API IDs auf andere IDs schliessen kann.

Wäre toll wenn ich ein paar Hinweise zu meinen Fragen bekommen könnte!

ricki-z, 2018-02-07:
Hallo Hannes,
country filter ist eingebaut: z.B. http://api.luftdaten.info/v1/filter/country=SE
Die IDs im JSON entsprechen den IDs im Archiv. Wahrscheinlich liest du die falsche ID aus. Im JSON sieht ein 'Datensatz' ungefähr so aus:
```
  {
    "id": 769339921,
    "sampling_rate": null,
    "location": {
      "longitude": "15.283",
      "country": "SI",
      "latitude": "46.316",
      "id": 3676,
      "altitude": "297.3"
    },
    "timestamp": "2018-02-07 21:58:04",
    "sensordatavalues": [
      {
        "value_type": "P1",
        "value": "120.30",
        "id": 1668820356
      },
      {
        "id": 1668820378,
        "value_type": "P2",
        "value": "61.15"
      }
    ],
    "sensor": {
      "id": 7269,
      "sensor_type": {
        "id": 14,
        "manufacturer": "Nova Fitness",
        "name": "SDS011"
      },
      "pin": "1"
    }
  },
```
und unter "sensor"->"id" steht die gleiche ID wie im Archiv.

Die NodeMCUs werden nie im Zusammenhang mit den Datenbank/Karten-IDs veröffentlicht. Damit wäre es möglich, mit einen WLAN-Scaner nach bestimmten Sensoren zu suchen. Die NodeMCU-ID ist Teil der MAC-Adresse der Sensoren (Hersteller-Präfix + ID in Hex).

ebner, 2018-02-11:
Hallo Rajko! Sehr gut, der country-Parameter war genau das wonach ich gesucht habe! Speichert ihr auch die Adresse bzw. Teile davon und macht sie über eine API verfügbar? Bei Airtube.info kann man nach Land und Ort filtrieren und ich würde gerne wissen ob der Ort irgendwo direkt verfügbar ist oder ob Airtube die Koordinaten selbst in Adressen aufgelöst hat, z.B. mit OpenStreetMap.

Wie funktioniert eigentlich http://api.luftdaten.info/v1/sensor/{apiID}/? Ich habe noch nicht herausgefunden wie ich zu einer funktionierenden API ID komme (habe alle möglichen IDs in http://api.luftdaten.info/static/v1/data.json getestet). Welche API ID brauche ich dafür?

ricki-z, 2018-02-18:
Die bulgarischen Sensoren werden dort zentral registriert. Dadurch sind den Leuten bei AirBG die Daten der bulgarischen Sensoren bekannt. Auf Grund der dort geltenden Gesetz sind alle Sensoren auch Eigentum der Foundation und die Paten stellen lediglich den Platz und Strom.
Und dadurch kennen die auch die Orte der Sensoren. Wobei es dank diverser offener APIs (z.B. Google) auch möglich ist, über die Geo-Koordinaten zumindest bis auf Stadtebene herunter passende Infos zu bekommen.

Auch wenn ich mir sicher bin, daß das irgendwo beschrieben ist: Die apiID ist die ID, unter der der Sensor auf der Karte zu finden ist. Und dies ist die ID, die mit der Bestätigungs-EMail gesendet wurde.
Wenn nicht explizit angegeben, arbeiten wir immer mit diesen Datenbank/API-IDs.

Im JSON sind dies die IDs direkt unter "sensor".

ebner, 2018-02-23:
Danke für die Erklärungen!

closed 2018-02-23 by ebner