opendata-stuttgart/meta #83: hackAIR leaks exact geolocation of luftdaten.info sensors

azrdev, 2018-03-01:
Apparently [hackAIR](https://platform.hackair.eu/) sources the luftdaten.info database for sensors. However, if you zoom in, you can see each sensor placed at an exact location on the map. This violates the promise given by luftdaten.info when requesting the adress of a sensor to keep it private, as stated [in the wiki](https://github.com/opendata-stuttgart/meta/wiki/Eintrag-in-unsere-Datenbank) and [the FAQ](https://luftdaten.info/faq/#toggle-id-6).

ricki-z, 2018-03-01:
They don't show the exact position of the sensors. There may be some sensors, where the position rounded to 3 decimals is near the exact position. 3 decimals correlates to a raster of around 100 meters.
Paint a raster, close your eyes and set random points into this raster. Then there will be points directly on intersections if you set enough points.
I've checked the hackair map. And I can't zoom in enough to see the exact position of a single sensor.

MkMunich, 2018-05-06:
Well, I checkt my sensor too and it is exactly the position ~ 5m deviation.

ricki-z, 2018-05-06:
@mkMunich And again: we use a raster. There may be some some sensors, where the position is exactly on the intersections of this raster. But you can't say which sensors.
Attached screenshots of the hackAir map for 2 of my sensors, the red points mark the exact position:
[image]
[image]