opendata-stuttgart/meta #84: Wifi Connection with a WPA2 EAP-PSK network

charliesz, 2018-03-06:
Hello everyone,

I try to connect my sensor to a network with WPA2 EAP-PSK (WPA2 Enterprise), so you do not have a single password, instead you have a personal user account with an individual username and a individual password. Neither I coud find a solution in the [wiki](https://github.com/opendata-stuttgart/meta/wiki/Konfiguration-der-Sensoren) nor in the _ext_def.h_ file.
I try to connect the sensor to the network of my university (RUB).
Here is an example for a proper connection file:
````
network={
    ssid="eduroam"
    scan_ssid=1
    key_mgmt=WPA-EAP
    eap=TTLS
    anonymous_identity="anonymous@ruhr-uni-bochum.de"

    # identity settings:
    identity="LOGINID@ruhr-uni-bochum.de"
    password="PASSWORD"

    # path to CA certificate:
    ca_cert="/etc/ssl/certs/Deutsche_Telekom_Root_CA_2.pem"

#    phase2="auth=PAP" # INSECURE
    phase2="auth=MSCHAPV2" # LESS INSECURE

    proto=RSN WPA
    mode=0
    pairwise=CCMP TKIP
    group=CCMP TKIP

    # check correct certificate
    subject_match="/C=DE/ST=Nordrhein-Westfalen/L=Bochum/O=Ruhr-Universitaet Bochum/OU=Network Operation Center/CN=radius.ruhr-uni-bochum.de"
}
````
[Information about the network](https://noc.rub.de/web/wlan)

I'm happy about any helpful advise or a link to another website/contact.

Thank you
Lukas

ricki-z, 2018-03-08:
Our sensor doesn't support WPA2 EAP-PSK. This needs too much resources on the esp8266 to run stable.

charliesz, 2018-03-08:
Thanks for your answer.
I just found those two promising Links in github for my problem [Link1](https://github.com/joostd/esp8266-eduroam) [Link2](https://github.com/esp8266/Arduino/issues/1032). I will update my progress as soon I have time to experiment with the new information.

L0laapk3, 2018-03-17:
@charliesz have you gotten anything to work without generating a certificate in advance with a pc?

charliesz, 2018-03-17:
Hey @L0laapk3 ,
so I'm absolutly new to networks, but if you mean being connected to "eduroam" without a certificate, yes kind of. In my android network settings I can ignore the certificate and still get access to the internet.

ricki-z, 2018-03-18:
The software needs to process the certificate given by the hotspot. And this needs too much RAM, even if we ignore it (this means no vaalidation is done).

ricki-z, 2018-03-27:
The ESP8266 and der Pi Zero use completely different cpus so there is no (easy) way to use our Arduino source code with 'only a few adaptions'. But there is a python script for Raspberries. You can find the link to this script here: http://luftdaten.info/faq/

closed 2018-04-17 by charliesz