opendata-stuttgart/meta #87: Temperatur Messung / Erwärmung Gehäuse

Clown007, 2018-03-24:
Ein tolles Projekt und die Erfassung des Feinstaubs funktioniert sehr gut.
Leider habe ich nicht darüber nachgedacht, dass die Wand an der der das Roht hängt von der Sonne beschienen wird. Heute ware es real 9 Grad und beide Sensoren haben 16 Grad (im inneren der Röhre) gemessen.

Wir habt ihr das gelöst?

Adorfer, 2018-03-26:
Ich hoffe ich verstehe die Frage richtig: Die Lösung besteht darin, "Feinstaubsensoren ohne Schlauch" zu nutzen, respektive diesen gar nicht aufzusetzen.
Nicht nur weil er technisch bei den gut durchlüfteten offenen HT-Rohren  überflüssig ist. Sondern weil die Schlauchlänge nach Monaten für Strömungswiderstand sorgt, wenn sich ein Staubfilm bildet.
Und dann den Temperatursensor (sinnvollerweise einen kompakteren BME280) direkt vor die Ansauglöcher des Sensors. Damit man Temperatur und Luftfeuchtigkeit des Feinstaubsensors bekommt. Und wenn es in der Glocke warm ist: Um so besser. Die Adiabatik hilft, genauere Messwerte zu bekommen, wenn dabei die Luftfeuchtigkeit sinkt.

Clown007, 2018-03-26:
Nach meinem Verständnis findet eben keine Belüftung des Rohres statt. Die 90 Grad Bögen sind nach unten geöffnet und werden in dem geraden Stück von der Sonne erwärmt. Ich messe mit dem BME280 (am Ansaugschlauch montiert) teilweise 10Grad höhere Temperaturen als  die Außentemperatur. In dem oberen Teil ist dann wahrscheinlich noch mehr. Ohne Ansaugschlauch messe ich imme diesselbe Luft, oder?

ricki-z, 2018-03-27:
Ja, es gibt keine wirkliche Durchlüftung des Rohres. Deswegen wird der Schlauch benötigt. Bei Verwendung der von uns vorgeschlagenen 87°-Ecken wird eine Durchlüftung relativ gut verhindert. Das soll auch dafür sorgen, das keine zusätzliche Luftströmung erzeugt wird, die dann die Luft vielleicht schneller durch den Sensor 'zieht' als die Software der SDS011 annimmt. Nur mit einem relativ konstanten Luftvolumen pro Zeiteinheit lässt sich aus den gezählten Partikeln eine Masse pro Kubikmeter abschätzen.

Clown007, 2018-03-27:
Vielen Dank für die Klarstellung. It's not a bug, it's a feature ;-)
Damit ist auch klar, dass die Temperatursensoren nicht für Außentemperatur Erfassung genutzt werden können. Das war meine ursprüngliche Idee.
Danke für das Projekt, es ist schon Klasse was mit kleinem Aufwand (durch eure Vorarbeiten) möglich ist.

Clown007, 2018-03-27:
Issue geschlossen Vielen Dank für die Klarstellung. It's not a bug, it's a feature ;-)
Damit ist auch klar, dass die Temperatursensoren nicht für Außentemperatur Erfassung genutzt werden können. Das war meine ursprüngliche Idee.
Danke für das Projekt, es ist schon Klasse was mit kleinem Aufwand (durch eure Vorarbeiten) möglich ist.

closed 2018-03-27 by Clown007