opendata-stuttgart/meta #88: API: How to filter area with negative lat / lon?

honkmaster, 2018-03-30:
How can is query / filter measurements values from the API with negative longitude or latitude values? If using a URL such as

https://api.luftdaten.info/v1/filter/area=34.226,-2.7213,200

the API returns an empty array.

ricki-z, 2018-03-30:
For the given area: there are no sensors ;-) . The format ist lat,lon,dist , so the coordinated are somewhere in Africa.
But there was also an error in the script. The parameter check refused negative values. This is now changed
The example for Bristol (UK): http://api.luftdaten.info/v1/filter/area=51.45,-2.583333,50

honkmaster, 2018-03-30:
> For the given area: there are no sensors ;-)

This was as intended (for privacy). I checked, it is working now. Thanks.

closed 2018-03-30 by honkmaster

honkmaster, 2018-03-30:
Somehow the API is now dead. 

[image]

ricki-z, 2018-03-30:
The database was defined with a max. ID of 2^31 values. This we have reached now. I'm changing the maximum to 2^63. But this needs some time.