opendata-stuttgart/meta #89: Schlauch "möglichst nicht transparent"

fpetry, 2018-04-24:
Warum ist die Transparenz genau relevant? Unglücklicherweise ist auf luftdaten.info neben dem Hinweis auch ein transparenter Schlauch abgebildet...

Transparenter Schlauch ist deutlich leichter zu beschaffen.
Vielleicht könnte man für einen passenden, nicht transparenten Schlauch auch eine Bezugsquelle angeben.

fpetry, 2018-04-25:
D.h. ein kurzes Stück als "Adapter" zum Sensor und dann einen Schlauch mit größerem Durchmesser um die Reibung zu mindern? Zu groß dann aber auch wiederum nicht, damit das Volumen ausreichend klein ist um noch Umgebungsluft abzukriegen.

Aber ok, das mit der Transparenz ist einleuchtend. Danke.

techniker24, 2018-06-21:
Der Mantel von einer typischen Mantelleitung (Gebäudeinstallation - NYM 3x1,5) ist nicht transparent und hat innen genau den passenden Durchmesser. Über die notwendige Länge kann man die Adern problemlos herausziehen, ohne den Mantel aufschneiden zu müssen. So erhält man einen Schlauch, der innen glatt ist. Man muss nur kontrollieren, ob keine Reste innen verblieben sind.