opendata-stuttgart/meta #90: Wiki translation

YaguraStation, 2018-05-30:
Since it's difficult to contribute to the translation of luftdaten.info, I thought it might be more productive and to start translating the GitHub Wiki pages. Whoever is in charge of the website can pick it up from the wiki, I guess. 

I started by linking the translation back and forth to the German original in this fashion: https://github.com/opendata-stuttgart/meta/wiki/EN-Assembly-of-parts 

Since GitHub Wiki does not have internationalisation and things are of course a bit messy with the German sidebar ... before continuing and maybe crossing any existing i18n plans, I wanted to be make sure this is OK with _y'all_. And I'll be happy to continue in this form until a smarter path is established. :blush:

YaguraStation, 2018-06-07:
Even if you're watching the repo, GitHub hasn't implemented notifications for Wiki changes :man_facepalming:. But you can follow them here: https://github.com/opendata-stuttgart/meta/wiki.atom

I've continued translation and will close this issue in a few days time if nobody has further comments.

ricki-z, 2018-06-09:
This way should be okay for all.

closed 2018-06-09 by YaguraStation