opendata-stuttgart/meta #91: Add license to meta

YaguraStation, 2018-06-07:
For other communities to reuse text, pictures, flyers, workshop handouts etc. safely, I recommend to add a license. I suggest CC BY-SA 4.0.

michael-k, 2018-06-08:
Why not [CC0](https://creativecommons.org/choose/zero/) (public domain)?

YaguraStation, 2018-06-08:
I suggested BY-SA because it is commonly used (eg. Wikipedia) and I don't have experience with CC0. Personally, I would be just as happy to do so. 

When comparing the two, I do find the _attribution_ with BY-SA compelling. Thereby the material itself could generate network effect.

Adorfer, 2018-06-14:
You can not enforce practically CC-BY-SA anyway (otherwise you will be singled out as an "Abmahnungs-Troll" violating the ethics of FOSS and ruining notion of the project in public etc.pp). 
And even if you put CC0, it's normal to get the credits anyway. It just gives (who like to respect and credit) others a bit more flexibility in doing so, in terms of layout. 

Or to put it the other way around: if you insist in CC-BY-SA i can put it on my checklist to find attribution violations and write emails for parts where i may have contributed.

YaguraStation, 2018-06-15:
Then lets do CC0, no problem. Will change PR. But this should include `/files`, because that's the idea, right? To also let pictures, PDFs etc be reused without worrying about violations.

michael-k, 2018-06-15:
You cannot include `files/*`, because none of the files were produced by any of us.  It might not legal for all of those files to be available via this repo.  [amtd-7-605-2014.pdf](https://github.com/opendata-stuttgart/meta/blob/master/files/amtd-7-605-2014.pdf) (as an example) already has a license (CC Attribution 3.0) and only the copyright holder has the right to republish it with another license.

YaguraStation, 2018-06-15:
Got it. How do we deal with this? Reproduce it, delete it, move it?