opendata-stuttgart/meta #95: Sensor verbindet sich nicht mit meinem Wlan

Jadoch48, 2018-07-07:
Hallo in die Runde,
ich habe mir einen Feinstaubdaten-Sensor gebaut und schaffe es einfach nicht, ihn mit meinem Wlan zu verbinden.

- wenn ich mit dem Handy einen Hotspot erstelle und im Sensor die entsprechenden Login-Daten eingebe, verbindet das Gerät korrekt und die Messdaten sind unter https://www.madavi.de/sensor/graph.php?sensor=esp8266-3871409-sds011 einsehbar. Im Handy taucht er als "ESP_3B12B1" bei den verbundenen Geräten auf.

- wenn ich in der Sensor-Konfiguration die Daten meines heimischen Wlans eingebe, erscheint der Sensor nicht mehr als Hotspot und im Router-Menü erscheint der Eintrag "unknown" bei den verbundenen Geräten. Allerdings erscheint er auch nach längerem Warten nicht unter https://www.madavi.de/sensor/graph.php. Auch unter dem Link http://feinstaubsensor-3871409.local/ kann ich nicht auf ihn zugreifen.

- Ich habe ein 2,4Ghz und ein 5Ghz Netz mit selber SSID. Der Router ist ein  Arris TG3442DE (Vodafone vermarktet ihn als Vodafone Docsis 3.1).

Könnt ihr mir helfen? Ich weiß nicht, wie ich das Problem weiter eingrenzen kann.

Danke und viele Grüße
Jakob

YaguraStation, 2018-07-07:
Two questions: Does your WLAN by any chance have a space in its SSID? Have you tried switching off 5Ghz on the AP temporarily to test?

You can review WLAN errors via serial monitor. See end of this article https://github.com/opendata-stuttgart/meta/wiki/Firmware-einspielen

Jadoch48, 2018-07-08:
Thanks for bringing up that article. No, there is no space in the wifi SSID. And yes, I tried swithing of 5Ghz with the same result.

I looked at the log via serial monitor and this is how the output reads out:

Any chance you can help me interpreting it? 

---- Result from Webconfig ----
WLANSSID: XXXXXXXXXX
----
Reading ...
PPD: 0
SDS: 1
PMS24: 0
PMS32: 0
HPM: 0
DHT: 1
DS18B20: 0
HTU21D: 0
BMP: 0
----
Send to ...
Dusti: 1
Madavi: 1
CSV: 0
----
Autoupdate: 1
Display: 0
LCD 1602: 0
Debug: 3
----
Restart needed ...

WiFi connected
IP address: 0.0.0.0
saving config...
{"current_lang":"DE","SOFTWARE_VERSION":"NRZ-2018-103","wlanssid":"XXXXXXXXXX","wlanpwd":"XXXXXXXXXX","www_username":"admin","www_password":"feinstaub","fs_ssid":"Feinstaubsensor-3871409","fs_pwd":"","www_basicauth_enabled":false,"dht_read":true,"htu21d_read":false,"ppd_read":false,"sds_read":true,"pms24_read":false,"pms32_read":false,"hpm_read":false,"bmp_read":false,"bmp280_read":false,"bme280_read":false,"ds18b20_read":false,"gps_read":false,"send2dusti":true,"send2madavi":true,"send2sensemap":false,"send2lora":false,"send2csv":false,"auto_update":true,"use_beta":false,"has_display":false,"has_lcd1602":false,"has_lcd1602_27":false,"has_lcd2004_27":false,"debug":"3","sending_intervall_ms":"145000","time_for_wifi_config":"600000","senseboxid":"","send2custom":false,"host_custom":"192.168.234.1","url_custom":"/data.php","port_custom":80,"user_custom":"","pwd_custom":"","send2influx":false,"host_influx":"api.luftdaten.info","url_influx":"/write?db=luftdaten","port_influx":8086,"user_influx":"luftdaten","pwd_influx":"info"}
Config written: {"current_lang":"DE","SOFTWARE_VERSION":"NRZ-2018-103","wlanssid":"XXXXXXXXXX","wlanpwd":"XXXXXXXXXX","www_username":"admin","www_password":"feinstaub","fs_ssid":"Feinstaubsensor-3871409","fs_pwd":"","www_basicauth_enabled":false,"dht_read":true,"htu21d_read":false,"ppd_read":false,"sds_read":true,"pms24_read":false,"pms32_read":false,"hpm_read":false,"bmp_read":false,"bmp280_read":false,"bme280_read":false,"ds18b20_read":false,"gps_read":false,"send2dusti":true,"send2madavi":true,"send2sensemap":false,"send2lora":false,"send2csv":false,"auto_update":true,"use_beta":false,"has_display":false,"has_lcd1602":false,"has_lcd1602_27":false,"has_lcd2004_27":false,"debug":"3","sending_intervall_ms":"145000","time_for_wifi_config":"600000","senseboxid":"","send2custom":false,"host_custom":"192.168.234.1","url_custom":"/data.php","port_custom":80,"user_custom":"","pwd_custom":"","send2influx":false,"host_influx":"api.luftdaten.info","url_influx":"/write?db=luftdaten","port_influx":8086,"user_influx":"luftdaten","pwd_influx":"info"}
?)⸮
⸮

⸮mounting FS...
mounted file system...
reading config file...
opened config file...
parsed json...
Starting Webserver... 0.0.0.0
6
Connecting to XXXXXXXXXX
........................................
Starting WiFiManager
AP ID: Feinstaubsensor-3871409
Password:

ricki-z, 2018-07-08:
Could you please send the characters used in the wifi password?
There should be a list of available wifi networks after soe seconds on the config page. Is your network listed there?

Jadoch48, 2018-07-25:
Sorry for the late response...

Yes, my network is listed on the config page. My wifi password contains some letters, two numbers, "!" and "_".

Jadoch48, 2018-07-26:
Ok so I renamed my two networks and changed the password but still was not able to connect.
I ordered a Freifunk router yesterday (a plan I have had for some time) and hope I will be able to connect to the Freifunk network once it is all set up.

ricki-z, 2018-07-26:
Have you checked your wifi router that there are no limitations (MAC filter, guest wifi settings, time limitations) for 'new' devices?

Jadoch48, 2018-08-11:
hi again,
yes, I checked all this :( With a rasperry pi I was able to connet to my router without any problems.
Anyway, I set up a Freifunk Router and was able to connect to the internet via the freifunk network.
So it's working now and I am happy :)
Thanks for the help
Jakob

Flole998, 2018-08-29:
You need a capture of the Wifi Messages to confirm, but I think it is most likely related to short guard Interval being enforced on the Router and the ESP lacking support for that.

bogomips, 2018-12-23:
hello, I have the same exact problem with the same exact model TG3442DE
My esp8266 perfectly connects to my phone (tethering)  while there is no way to connect to my Vodafone router.
For two days I am trying to find the root cause, using Wireshark etc... no way.
Does someone know what the problem is? (I don't wanna buy a new router!)
Thanks!

Flole998, 2018-12-23:
> I think it is most likely related to short guard

As I said before.... What does the Wireshark capture say?

ricki-z, 2018-12-23:
The sensor only supports 2.4 GHz wifi. And the sensor seems to prefer 811.2n (my Fritz!Box is set to 811.2b+g+n).