opendata-stuttgart/meta #98: Since firmware NRZ-2018-119A no data written to local InfluxDB

SmartBulb67, 2018-11-21:
Hi there,
I like your project very much and have already set up two sensors at different locations, which provide data since some months. They have the IDs 242418 and 4506372.
Both sensors also write data into different InfluxDBs.
One of the sensors (242418) stopped the data input into the InfluxDB exactly 2018-11-19 08:10 and one day later I realized the sensor has a new firmware (NRZ-2018-119A).

How could I check, if this firmware has something to do with the issue?
I could not find any debug messages in the web GUI.
Rebooting the sensor and re-setting the configuration I have already tried.

Any more ideas?

Best regards
Daniel

ricki-z, 2018-11-21:
@SmartBulb67 please move this issue to sensors-software where it belongs to.

SmartBulb67, 2018-11-21:
Yes, you are right. 
Unfortunately I have no idea how to perform this task?
Should I close and re-open? Or is there a more elegant procedure for this?