opendata-stuttgart/meta #99: Switch English shopping list links to English AliExpress

raivisdejus, 2018-12-04:
Hi,

Current English How to page is good, but it could be better if the links to shopping list items would lead to English AliExpress as for us non-German speakers it is hard co complete the checkout in Ali express and changing the language the search term is lost.

YaguraStation, 2018-12-04:
I'm assuming you're talking about the website? In the Wiki I've tried to make it language-independent: https://github.com/opendata-stuttgart/meta/wiki/EN-Components-supply

raivisdejus, 2018-12-05:
Yes I am talking about the website https://luftdaten.info/en/construction-manual/
This is a small issue, but important one, should be easy to fix.

If necessary I am willing to help with the fix

ricki-z, 2018-12-05:
Hi @raivisdejus, could you please check the links? I have changed them and the should go now to the english version of AliExpress. The site is setting a language cookie, so you should go to AliExpress directly and switch to english before you test this.

raivisdejus, 2018-12-06:
@ricki-z Excellent, works like a charm :)

closed 2018-12-06 by raivisdejus