opendata-stuttgart/meta #102: Why my nodemcu reboot when refresf the page during a measurement (bug or normal?) :

MagTun, 2018-12-09:
I am not sure if this is the real cause, but when I refresh the page http://IP/values  by pressing F5 just when a measure is done, then the nodemcu bugs and it reboot itself

Step to reproduce: 

1. open http://_the_local_IP/values 
2. wait until you see 143 seconds since last measurement.
3. press several times per second `F5`(just to be sure to press on time)
4. if you see the next measurement update, continue to press  `F5` a couple of more times

At this point, the browser returns the error `can't connect`. If I wait for several seconds (maybe a minute) then I see the page without any measure and with the red warning "Still 143 seconds until first measurement."

I am using chrome. 

Is it normal?

YaguraStation, 2018-12-09:
During readings the web UI won't be available. But after that it should show the current readings. So, it's not normal. 

Is it reporting correctly to the API?

MagTun, 2018-12-09:
I don't know how I could check if it's reporting correctly to the API. Do you mean madavi API or luftdaten API (I am not yet registered with luftdaten).

MagTun, 2018-12-09:
Actually, I think it's not reporting to madavi because one time, just before getting the bug, I saw the new measure values. Later that day, I downloaded the csv from madavi and these values weren't in the csv. The previous one and the next one where in the csv though. 

I have seen that there is a debug feature in the configuration. I have set it to "high info", but I don't know how I cant get the log.