opendata-stuttgart/meta #105: English sensor registrations

raivisdejus, 2019-01-14:
Is everything ok with sensor registrations on [English page](https://luftdaten.info/en/construction-manual/)?
A while ago I submitted registration for esp8266-1519829 and esp8266-13058714, but the submissions do not seem to be processed.

You seem to be using WordPress with Contact Form 7. With [Flamingo plugin](https://wordpress.org/plugins/flamingo/) you can save all form submissions in WordPress. This is a good backup if there are some issues with delivering e-mails. 

If there is any way I can help, let me know. Would be glad to help this great and useful project :)

raivisdejus, 2019-02-07:
All good, thanks!

Got responses from twitter and all my sensors are now registered. 

An idea for future - add a note on the website if the registrations are expected to take longer.

closed 2019-02-07 by raivisdejus

ropeters68, 2019-02-07:
Lucky you!👍(still waiting)