opendata-stuttgart/meta #106: Archivdaten vom 21.11.2018 fehlen

shred, 2019-01-23:
Siehe: https://archive.luftdaten.info/2018-11-21/

Leider fehlen die Daten für den Tag auch im `csv_per_month`-Ordner.

shred, 2019-05-20:
Gibt es Neuigkeiten hierzu? Sind die Daten noch restaurierbar?

Bouni, 2019-08-23:
Es scheint das am 16. August irendwas kaput gegangen ist, seit diesem Tag gibts nichts neues mehr unter archive.luftdaten.info

ricki-z, 2019-08-26:
Die Daten ab dem 16. August sollten in den nächsten Tagen nachkommen und dann auch wieder aktuell gehalten werden.
Es gab ein kleines Performanceproblem auf dem betreffenden Server, so daß die Archivierung temporär ausgesetzt wurde. Naja, und das ganze natürlich während der Urlaubszeit ...

shred, 2019-09-23:
Es wäre die einzige Lücke in meiner eigenen Datenbank für meinen Sensor. Aber eigentlich ist es nur ein Schönheitsfehler, kein echtes Problem. Dafür lohnt sich ein größerer Aufwand nicht.

Da sich hier sonst niemand deshalb gemeldet hat, schließe ich den Bug von meiner Seite.

closed 2019-09-23 by shred