opendata-stuttgart/meta #107: Configure GY-BM EP280 sensor

polvonbraun, 2019-01-27:
Hallo,

Mein ester Sensor ist schon seid einiger Zeit online, nun wollte ich noch einen zweiten zusammen bauen und an einer Anderen Lokalität aufbauen.

Für den ersten Sensor verwende ich zusätzlich einen DHT22 Chip.
Für meinen Zweiten habe ich nur einen GY-BM EP280 Chip gekauft.

Ich habe nun alles zusammengebaut und die Firmeware geflashed. Es scheint soweit auch zu klappen, der Sensor liefert Daten zum Feinstaub: https://www.madavi.de/sensor/graph.php?sensor=esp8266-9001232-sds011
Es tauchen allerdings keine Werte zum BME280 auf...

Hier in der [Anleitung ](https://luftdaten.info/faq/#toggle-id-14) steht: "Nach den Anschließen müssen die Sensoren noch in der Konfiguration aktiviert werden." Allerdings habe ich keine Ahnung was das genau bedeuten soll... Wenn ich mich zum konfigurieren mit dem NodeMCU verbinde finde ich dort nichts zum "konfigurieren" außer WiFi.

Hoffe jemand kann mir weiter helfen, ich habe mich an die Infos in der  [Anleitung ](https://luftdaten.info/faq/#toggle-id-14) gehalten.
Vieleicht habe ich etwas bezüglich I2C falsch gemacht, damit kenne ich mich nicht wirklich aus... GND des BME280  habe ich an den gleichen wie sds011 angeschlossen.

Hier ein Paar Bilder wie das ganze im Moment bei mir aussieht:

[image]
[image]
[image]
[image]

[image]
[image]

ropeters68, 2019-01-27:
A few releases ago a smaller initial logon screen was introduced. In this sceen you should fill in the network credentials and save/restart . To change the sensor settings you will need to logon a second time by entering the IP address of the sensor in your browser (the IP address is generated by your wifi router), then you will get the full configuration page.

polvonbraun, 2019-01-27:
@ropeters68 thank you very much, this was the decisive tip I needed.

the sensor is working nor and logging data from the BME 280!

[image]

ricki-z, 2019-01-29:
@polvonbraun can you close this issue?

closed 2019-01-29 by polvonbraun