opendata-stuttgart/meta #108: A working LCD ?

mgaida11, 2019-02-01:
Hello, 

would you please give a suggestion, which LCD (1602 or 2004) works / should be bought?
I use a LCD2004 with 5V and I2C but all I can see are two rows of blocks. 
An OLED-Display works fine but it's too small for me ;-)

Thanks
MIchael

ricki-z, 2019-02-01:
Please check the contrast ...

mgaida11, 2019-02-01:
Already done. I've been through 4 or 5 2004-LCD already. It's strange but I'm happy to find one, that works. It is labeled "J204A". The other LCD labeled "2004A" and "2004" (all with I2C and similarily looking) won't work. 

Now I'm fighting with unusually high PM2.5- and PM10-values, all around 240-260. Next to the device I have a working one, that delivers 1/10 of that. Do you have an idea what causes that?

ricki-z, 2019-02-01:
This is normally a voltage problem. If the voltage goes below 4.8V the PM values become too high. SO check the power supply, the cables and connectors.

mgaida11, 2019-02-01:
Really strange. As I said before, I have got here two devices. One's running fine, the other is a problem.
The good one works on a NodeMCU V3. It gives the incoming 5,1V to the sensor.
The bad one works on an NodeMCU V2. It gives only 4,7V to the sensor.

Changing the cables makes no difference.

When I cange the sensors to the the NodeMCUs, I get with the NodeMCU V2 good readings (around 10ug).
The "problem-sensor on the NodeMCU v3 with 5,1V gives now 110-120ug (before: 240).

The result for me:
- the new sensor reacts extremely to V-differences and
- it is somehow broken, because it also gives too high reading.

- the old sensor is much robust to V-differences.

What do you think about this strange behaviour?

mgaida11, 2019-02-01:
Flashed a D1Mini that was lying around with the firmware. It delivers 4,75V to the sensor. Its results are between 220-240 ug, as before.

ropeters68, 2019-02-01:
I had the same issue with one sensor, I just let it run to see what would happen. After about 8 hours it went back to normal. I cheched the values with two other sensors and no significant differences.

mgaida11, 2019-02-02:
1300 times and no change :-(

ricki-z, 2019-02-02:
And again: the SDS011 needs minimum 4.8V. Why do you think that the values will change if your board only delivers 4.75V?
Check your power supply. It should be able to deliver 1 Watt. Even if this is not needed the most time, there may be situations where the sensor may need this. I.e. if the wifi part and the SDS011 are active. The SDS011 alone may need 0.5 Watt (5V, 100mA).

freitux, 2019-02-03:
Am Freitag, 1. Februar 2019, 20:22:39 CET schrieb Rajko Zschiegner:
> Please check the contrast ...