opendata-stuttgart/meta #111: API returns error 403: Forbidden

pmdstudio, 2019-02-21:
Hi there,
I am trying to load data for selected country and use the filter URL: http://api.luftdaten.info/v1/filter/type=SDS011&country=BG. It works from the browser and on all the test sites I've tried. But when try to open the URL from my site (www.vremetodnes.bg) the API returns ERROR 403: Forbidden. The response is:

<html><head>
<title>403 Forbidden</title>
</head><body>
<h1>Forbidden</h1>
<p>You don't have permission to access /v2/now.filtered.pl
on this server.<br />
</p>
</body></html>

All the other API URLs are accessible from my server: 
http://api.luftdaten.info/static/v1/data.json
http://api.luftdaten.info/static/v2/data.json
etc...

My question is why is that and how I can fix it?

Thanks,

ricki-z, 2019-02-21:
This server is making request like '/v1/filter/type=SDS011&area=43.1667,23.8000,3' at a speed of up to 10 requests per second. Sorry, but this server isn't meant to serve only one user.
If you want to use this data at this speed then please copy the data.json and write your own filter. Then you can can see how many resources are needed by this.

pmdstudio, 2019-02-21:
> This server is making request like '/v1/filter/type=SDS011&area=43.1667,23.8000,3' at a speed of up to 10 requests per second. Sorry, but this server isn't meant to serve only one user.
> If you want [...]

Hi. The speed is ok. I don't need more. I understand that this is an open source project and you can use it as it is with all the limitations. My question was what's the reason one of my sites (vremetodnes.bg) to be blocked but all other sites can make requests to the API? I am trying to figure out if I am making any mistakes in the request. While I was developing the production site I was using a development server and I did a lot of requests from other domain and everything was good. Once I've uploaded the code to the production site the domain was blocked immediately. And I am curious why?

ricki-z, 2019-02-21:
Just to be clear: 10 requests per second from one IP ist far too much!
Actually we have more than 6000 users. If only 1 percent of them would do the same like you we would have 600 requests per second. With 0.2 seconds CPU time per request this would be 30 seconds CPU time per second only for these requests. You can try to find a server with 32 CPU cores and look at the price of such a system. and we would need a larger system as we are still growing and have to serve other requests also.
This is why we block such systems as soon as we detect them. Maybe your development system wasn't detected jet.
I also don't know if there are any other sites on the same system / IP that may have caused the block of this IP.

pmdstudio, 2019-02-22:
> Just to be clear: 10 requests per second from one IP ist far too much!
> Actually we have more than 6000 users. If only 1 percent of them would do the same like you we would have 600 requests per [...]

So to summarize (correct me if I am wrong):
1. API can block IP's if there are too many request. It blocks the IP your site is coming form.  
2. For further usage If you want to make more request to the API it's better to use the static json file - copy the data.json file and filter the sensors by yourself

Currently I am making 2-3 requests per day, but my IP is blocked maybde bacause if I am using shared hosting and some other sites make more requests.

I will make change my code to use the static json file.

Thanks for the responses.

ricki-z, 2019-02-22:
Your summary is correct.
We can only see the IP address of the systems that access the API (there is no http header stating the 'hostname'). So we can only block offensive IP addresses. This may be a problem for shared hosted sites, but we don't have any other chance to avoid an overload of our system.
To your system: the IP address of your site www.vremetodnes.bg is 164.138.216.124. Some of the log entries:
164.138.216.124 - - [01/Feb/2019:05:11:21 +0100] "GET /v1/filter/type=SDS011&area=42.6983,23.3199,10 HTTP/1.0" 403 393 35100 "-" "-" - -
164.138.216.124 - - [01/Feb/2019:05:11:21 +0100] "GET /v1/filter/type=SDS011&area=42.6983,23.3199,3 HTTP/1.0" 403 393 34666 "-" "-" - -
164.138.216.124 - - [01/Feb/2019:05:11:22 +0100] "GET /v1/filter/type=SDS011&area=43.6667,28.2833,10 HTTP/1.0" 403 393 34599 "-" "-" - -
164.138.216.124 - - [01/Feb/2019:05:11:22 +0100] "GET /v1/filter/type=SDS011&area=43.6667,28.2833,3 HTTP/1.0" 403 393 34425 "-" "-" - -
So yes, it looks like another website is accessing our API from this IP as the requests are completely different from those mentioned in your first post.

closed 2019-12-16 by pmdstudio