opendata-stuttgart/meta #112: Was sind Samples?

ituri, 2019-03-17:
Hallo,

leider konnte ich in der Wiki keine Erläuterung finden, deshalb frage ich hier und füge natürlich auch gerne selber eine Erklärung ins Wiki ein: Wenn ich die Daten meines Sensors in einer InfluxDB speichere, wird dort auch der Wert "Samples" abgespeichert. Was hat es damit genau auf sich und in welcher Maßeinheit wird dieser Wert gemessen?

Vielen Dank!

ricki-z, 2019-03-17:
samples ist die Anzahl der Durchläufe der Hauptschleife. Es trifft hauptsächlich Sensoren, die bei jedem Schleifendurchlauf ausgelesen werden (z.B. PPD42NS, später auch Lärm). Hier muss teilweise auch bekannt sein, wie oft diese ausgelesen werden können.
Ansonsten nutzen wir dies als einen der Indikatoren für Performance-Optimierungen.
Das ganze ist einheitenlos.

ituri, 2019-03-22:
Danke!

closed 2019-03-22 by ituri