opendata-stuttgart/meta #113: Lebt das Projekt "Luftdaten.info / Feinstaubsensor" überhaupt noch?

iceweasel1, 2019-04-15:
Hallo zusammen,

die gezielte Frage steht bereits in der Überschrift. Ich habe mich das gesamte Wochenende stundenlang durch luftdaten.info, hier in GitHub, OK Lab Stuttgart (Facebook), usw. usw. usw. durchgelesen. Ob die Termine (jeden 2. Dienstag im Monat) überhaupt noch stattfinden; für mich ein Rätsel. Sei es hier oder auf der Webseite, dort werden URL-Links bezüglich der Anmeldung für das "jeden 2. Dienstag im Monat" angegeben die z. B. auf Teilnehmerlisten zeigen, wie bspw. Doodle, TU-Dresden, Meetup. Alle samt sind weder aktuell noch irgendwie richtig funktionsfähig. Sämtliche der genannten Doodle, TU....etc. haben Daten drin, die von 2016/2017 stammen und ebenso keine neue Anmeldung zulassen, oder darauf hinweisen. shackspace hat im eigenen Kalender zum Feinstaubtreffen-Bau nicht den übernächsten Dienstag (was der 23. April wäre) stehen, sondern erst den 14. Mai 2019. Doch nicht jeden 2. Dienstag im Monat?

Sonstige Blog/Textbeiträge sind fast leider auch weit in das Jahr 2018 zurück. Eine Eintragung für die öffentliche Karte, auf der Webseite wird ja angegeben das es etwas dauern kann, dauern laut einigen aktuelleren Kommentaren, die ich so gefunden habe, anscheinend rund 1 Monat, teils aufwärts. Echt?

Mein Interesse an dem Projekt ist groß. 2 Feinstaubsensoren habe ich jetzt zusammengebaut. Eigentlich wollte ich heute die Bauteile für 8 weitere kaufen. Mich beschleicht leider das Gefühl, das es hier wie erwähnt nicht weiter geht. Da werde ich wohl mal keine 8 weiteren Zusammenbauen.

Auch gibt es etliche widersprüchliche Angaben/Aussagen zu gewissen Bauteilen, die hier in GitHub angegeben werden, ein Betrieb (Firmware) damit möglich wäre; jedoch auf der Webseite selbst in der FAQ ein Teil davon erwähnt wird und ein anderer überhaupt nicht. Eines zum Beispiel davon wäre der BME280 (Gas). Wird hier angegeben, kein Wort auf der Webseite auf einmal. Bevor ich mir 10 von den Teilen bestelle und es dann, widersprüchlich, doch nicht geht, lasse ich es mit dem Bauteil vorerst.

Ich hoffe doch das es weitergeht. In meinem Landkreis gibt es bislang 5 Stationen, die mir angezeigt werden. Drei davon, meiner Einschätzung nach, müssten vor kurzem erst dazugekommen sein. 2 Stationen hatte ich schon vor langer Zeit (ohne die drei weiteren) gesehen. Mit meinen 2 fertigen (noch nicht eingetragen), plus den eigentlich 8 weiteren geplanten Stationen, würde ich gerne im meinem Landkreis an unser am meisten befahrenen/durchfahrenen Straße (Bundesstraße) diese anbringen. Ein Landkreis neben meinem Landkreis hat laut Karte noch keinen einzigen. Das wäre mein Interesse ebenso groß, dort einige hinzustellen. 120 km hätte ich nach Stuttgart. Umsonst für einen nicht stattfindenden Termin will ich jedoch nicht hinfahren. Zu Schwätzen würde es vieles geben.

Grüße
René

ricki-z, 2019-04-15:
Hallo,
zu den Terminen: Laut den Protokollen hier auf Github (https://github.com/opendata-stuttgart/meta/wiki in der rechten Spalte) haben die letzten Termine am 26.3. in der Stadtbibliothek und am 9.4. im shackspace stattgefunden und es haben sogar 7 bzw. 13 Leute daran teilgenommen.
Einer der Temin-Links führt auf Meetup, funktioniert und auch die Termine sind aktuell. Der dortige Link zum Doodle der TU Dresden dient wie im Text beschrieben zur Anmeldung zum Sensorbasteln. Dieser hat keine aktuellen Termine, weil wir gerade keine Teile da haben.
Es gibt in jedem Monat nur exakt einen 2. Dienstag. Der 23.4. ist der 4. Dienstag im Monat, an diesem Tag findet das Treffen in der Stadtbibliothek statt. Deshalb gibt es dort auch keinen Termin auf der shackspace-Seite.
Der letzte Blogeintrag stammt vom 28.2. 2019, die davor vom 15. Oktober 2018 und vom 25. September 2018. Ob dies jetzt "weit in das Jahr 2018 zurück" ist, kommt wohl auf das persönliche Zeitgefühl an. Der letzte Tweet auf twitter.com ist von gestern, der letzte Eintrag auf Facebook vom 9.4. 
Und wenn man hier auf Github mal genauer hinschaut, sieht man, daß gerade an "meine.luftdaten.info" gearbeitet wird. Diese Seite soll es ermöglichen, die Sensoren selbst anzumelden. Leider machen wir das Projekt alle ehrenamtlich, organisieren, programmieren und entwickeln das also alles in unserer begrenzten Freizeit. Deshalb geht vielleicht manches nicht ganz so schnell, wie das hier offensichtlich von einigen Nutzern erwartet wird.
Die Seite https://meine.luftdaten.info/ ist gerade noch in der Beta-Test-Phase. Schaue dir doch einfach mal die Seite an, vielleicht willst du ja die noch von dir fehlenden Sensoren selbst anlegen. Es sollte eigentlich soweit alles funktionieren. Im Moment kann die Seite aber manchmal kurzfristig nicht erreichbar sein, wenn wir daran arbeiten.
Zu den anderen Sensoren wie z.B. dem BME280: In den FAQ findet unter dem Punkt "Andere Sensoren anschließen" eine Beschreibung, wie der BMP180, der BMP280 und der BME280 angeschlossen werden und was danach noch zu tun ist.

iceweasel1, 2019-04-15:
@ricki-z Danke für deine rasche Antwort. Vorweg, ich wollte keinem irgendwie zu nahetreten. Sollte es so rübergekommen sein, bitte ich dies zu entschuldigen. Ich respektiere das Ehrenamt ungemein und habe davor, gerade auch was dieses Projekt ausgibt, einen Heiden Respekt davor. Die Erwartung, dass es hier schnell(er) gehen sollte/könnte wollte ich wirklich nicht damit rüberbringen. Es hatten sich letztlich viele Faktoren zusammengetragen, die mich verwundern ließen und ich den Post hier absetzte.

Ich war, wie man dem Text entnehmen konnte, schlichtweg verwundert, da es an für "mögliche Neue" irgendwie still ist, bzw. einige Fragen offenbleiben. Twitter ja, da ist aktuell das Ganze. Facebook, du meinst 09.04.2018? den angepinnten Beitrag?

Merke gerade, dass ich den falschen Sensor geschrieben hatte. Ich meinte den Gas-Sensor MQ135. Der BME280 wird wie du erwähnt hast, angegeben in der Anleitung, das stimmt. Der Gas jedoch nicht; wiederum aber hier in GitHub bei der Bauteilliste.

Das mit "Jeden 2. Dienstag im Monat" habe ich dann wohl fälschlicherweise anders interpretiert, dass es jeden zweiten Dienstag im Monat wäre. Führt etwas zur Verwirrung das Wort "jeden". Vorschlag: Wäre "Am 2. Dienstag jeden Monats" besser?

Wunderbar die meine.luftdaten.info! Habe mich eben registriert. Danke für den Tipp. Werde dort nach und nach die Sensoren eintragen die ich fertig habe. Zwei Fragen hierzu. Ein Upload-Feld eines Fotos für eure Interna fällt bewusst weg? Bei den Sensoren, zwei kann man auswählen. Wie ist das, wenn ich den SDS011, mit dem BME280 nehme und zusätzlich den MQ135 drannehme (vorausgesetzt das man das so anschließen könnte), die Auswahl eines dritten Sensors?

Ansonsten, dickes Danke für alles!

ricki-z, 2019-04-15:
Bei mir kommt auf Facebook als erster Beitrag einer zu "DUST | VR Triptych | 2018/19" vom März diesen Jahres. Aber da "spinnt" auch Facebook manchmal. Bei meinem eigenen Facebook-Account sehe ich gerade nur noch 3 oder 4 Beiträge, wenn ich nach Zeit sortiere.
Beim MQ135 hat wohl jemand etwas falsch ins Wiki eingetragen. Dieser Sensor wird nicht unterstützt. Leider kann das Wiki von fast jedem editiert werden. Ich nehme das gleich raus.
Und bei meine.luftdaten.info wollen wir vor allem erst einmal die 95% der Nutzer erreichen, die nur 2 Sensoren verbauen (einmal Feinstaub, einmal Temperatur/Luftfeuchte/Luftdruck). Das 3 und mehr verbaut sind, kommt eher selten vor. Aber auch das ist in Arbeit.
Das Foto wollten wir bisher haben, um ein bißchen kontrollieren zu können, daß der Sensor richtig aufgehängt wird. Und welche Lösungen die Bastler bezüglich der Aufhängung finden. Da die Sensoren fast nie wirklich falsch aufgehängt wurden, ist das Foto inzwischen nicht mehr wirklich nötig.

ituri, 2019-04-15:
Danke an @ricki-z für die Antwort!
Auch mich trieben, wie schön an anderer Stelle geschrieben, ähnliche Gedanken um. Einerseits ist das Projekt durch Artikel in den Medien etc. ziemlich präsent in der öffentlichen Wahrnehmung, andererseits erweist sich die Recherche zu etwas tiefergehende Fragen oftmals als Sackgasse. 
Komme zwar nicht aus Stuttgart, aber wenn ihr Unterstützung in Projektbereichen benötigt, die keine Programmierkenntnisse erfordern, würde ich gerne aushelfen.

iceweasel1, 2019-04-15:
@ricki-z Vielen Dank für deine Antworten. Das wichtigste ist hiermit beantwortet worden. Wunderbar. Ein Glück das ich den MQ135 nicht schon bestellt hatte. Das ergibt Sinn, mit in der Regel 2 verbauten Sensoren. Somit erübrigt sich, aufgrund des MQ135, die Option eines dritten anzugeben. Schön mit der Fotosache, wenn das so gut wie immer funktioniert. ;-)

@ituri Danke für deine zusammengefasste Aussage.

Ich schließe mich @ituri an was die Unterstützung betrifft. Bis auf die Codes schreiben, stehe ich gerne bei den anderen Dingen zur Verfügung, wenn man helfen kann. Ich werde nach Stuttgart beim nächsten Treffen einfach mal kommen.