opendata-stuttgart/meta #119: BME280 keine Daten mehr nach ca. 1-2 Tagen

FreeTechNick, 2019-08-11:


FreeTechNick, 2019-08-11:
Hallo zusammen,

ich habe seit mehr als einem Jahr einen Feinstaubsensor in Betrieb. Diese läuft auch sehr gut. Jetzt habe ich vor ein paar Monaten einen BME280 Sensor hinzugefügt, dieser liefert auch meistens 1-2 Tage Daten. Dann startet scheinbar der Feinstaubsensor neu und der BME280 sendet keine Daten mehr, die anderen Sensoren senden die Daten weiter. Erst, wenn ich den Feinstaubsensor vom Strom trenne und damit völlig neu starte liefert der BME280 wieder Daten. Gibt es hierzu eine Lösung eine Tipp? Danke.

Viele Grüße Markus

computerfreund03, 2019-08-22:
Hatte das gleiche Problem, mit dem DHT22. Anderes Netzteil hat geholfen. Problem ist der Power Cycle.