opendata-stuttgart/meta #121: Viel zu hohe Werte

jonashaag, 2019-10-11:
Ich hoffe, das ist hier nicht der falsche Ort für Support. 

Ich habe jetzt mal so einen sensor zusammengebaut entsprechend offizieller Empfehlung. Hängt jetzt außen am Balkongeländer, ca 5 cm von der Hauswand entfernt. 

Das Ding misst in der Größenordnung 400 µg/m³. Trotz befahrener Straße erscheint mir das etwas zu hoch 😬 was könnte ich falsch gemacht haben?

Danke!

ricki-z, 2019-10-11:
Hallo,
ich hoffe ich kann dir weiterhelfen. Kannst du bitte prüfen, ob die gleichen Werte auch mit einem stärkeren Netzteil auftreten? Das wäre der häufigste Grund für so hohe Werte.
Der 2. wäre ein verunreinigter SDS011, den ich aber bei einem neuen Sensor mal ausschließe.
Und als drittes könnte man noch einmal die Steckverbindungen und Kabel prüfen.
Beim 1. und 3. Grund wird zuviel gemessen, weil die anliegende Spannung bzw. der verfügbare Strom nicht genug ist und dann der Laser minimal dunkler ist. Das wird beim SDS011 auf Grund der optischen Messung dann als Verdunklung durch Partikel gewertet und gezählt.

jonashaag, 2019-10-11:
Danke für die Tipps! Netzteil ist Apple 1A bei 5V. Ich schaue mal, ob ich ein stärkeres da habe und prüfe auch die Kabelverbindungen.

jonashaag, 2019-10-11:
Also, was nicht funktioniert hat:

- Netzteil mit 1,5A
- Schlauch abmachen
- Alles neu zusammenstecken

Interessant ist aber, dass an Vin/Gnd (v2 NodeMCU) höchstens 4,54V Spannung anliegen. Am USB-Netzteil habe ich 5,05V gemessen. Im SDS011 Data Sheet steht, dass er mindestens 4,7V braucht.

ricki-z, 2019-10-11:
Die zu niedrige Spannung dürfte das Problem sein. Kannst du eventuell ein anderes USB-Kabel nutzen?
Bei der NodeMCU V2 sollte eigentlich nicht viel zwischen USB-Eingang und den Pins Vin/GND sein, was einen Spannungsabfall erklären könnte.

Adorfer, 2019-10-15:
Also entweder einen 5V-Regler (Buck-Converter) direkt auf den ESP stecken und mit 12V speisen. 
(das dürfte da technisch Sinnvollste sein)

Alternativ USB-Netzteil mit 5,5V kaufen (gibt's auch)
Oder Netzteilkabel auf möglichst der kompletten Strecke gegen 2,5mm^2 Lautsprecherlitze tauschen.

jonashaag, 2019-10-15:
Ich werde es mit einem 5,5V Netzteil versuchen. 12V Netzteil dürfte entweder den NodeMCU oder den SDS011 beschädigen (10V Max Spannung und 5,7V Max Spannung wenn ich es richtig verstanden habe).

Adorfer, 2019-10-16:
Welcher Teil des Satzes
 
> einen 5V-Regler (Buck-Converter) direkt auf den ESP stecken und mit 12V speisen.

war unklar?
Ich setze natürlich voraus, dass der Buck-Converter nicht defekt ist.

jonashaag, 2019-10-16:
Ups. Ich habe „oder“ gelesen.