opendata-stuttgart/meta #123: Ausgerolltes Firmware-Update führt zu massiven Probleme mit Sensor & Messung

iceweasel1, 2019-11-13:
Hallo nochmals,

ich hatte bereits vor 12 Tage einen Hinweis und Fragen [hier ](https://github.com/opendata-stuttgart/meta/issues/122) gestellt. Die Problematik ist mehr geworden als mir lieb ist. Aus den Socken haut es mich, als ich eben die E-Mail (vermute automatisiert versendet) erhalten habe, das mein Sensor angeblich seit mehr als 24 Stunden keine Daten mehr in die Datenbank gesendet hätte. Die sofortige Überprüfung ergab, das es eine 50/50 Sache ist. 

Seit gestern (12.11.2019) nach 07.03 Uhr sind bis jetzt keine PM-Daten mehr zu sehen. Durch Grafana kann ich es auf diese Uhrzeit eingrenzen. Was jedoch keinen 24h-Ausfall anzeigt, ist das WLAN-Signal und die BME280, also Temperatur, Luftfeuchte und Luftdruck. Diese Daten sind bis jetzt soweit im Archiv und Grafana zu sehen.

Was ich im anderen Thread vor 12 Tagen beschrieben hatte, war, das seit ein neues Firmware-Update (automatisiert) ausgespielt wurde, es meinem gebauten Sensor letzten Endes und mittlerweile massive Probleme bereitet. Die Sache, die ich erwähnt hatte, das die PM2.5 & 10-Angabe in der Übersicht, wenn man diesen im Netzwerk über die IP-Adresse ansteuert, leider oft leer bleiben/keine Zahlen mehr, sondern Bindestriche anzeigen. Das hat sich die vergangenen 12 Tage sehr oft wiederholt, bis jetzt, das es gar nichts mehr anzeigt. 

Da die E-Mail mit dem Hinweis der "... mehr als 24 Stunde..." jetzt auch noch kam, bestätigt sich der Verdacht und ebenso die Beobachtung der letzten 12 Tage, dass, sobald es einfach diese Bindestriche anzeigt, eine Übertragung in die Datenbank zu 50 % stattfindet und ebenso 50 % nicht stattfindet. Dies konnte ich oft beobachten, das mal bei angezeigten Bindestrichen die Daten dennoch eingespielt wurden, und leider wieder nicht. Im Archiv dann von der notierten Uhrzeit bei der Nachschau zu überprüfen gewesen. 

Das Thema, dass ebenfalls angesprochen wurde, das bei "Anzahl Messungen" es nach mehreren hundert, wieder und wieder mal bei 0 anfängt, obwohl kein manueller oder automatisierter Neustart ausgeführt wurde, besteht weiterhin. 

Ich führe die gesamten genannten aufgetretenen Probleme auf das ausgerollte Firmware-Update raus. Warum auch immer, aber anscheinend verträgt es mein nach der Anleitung gebauter Sensor nicht. Seit meinem Beginn April bis zum vor kurzem Update, gab es kein einziges Problem bzw. Probleme mit der Firmware. 

Das war das vorherige Thread: https://github.com/opendata-stuttgart/meta/issues/122

Kommt da ein gefixtes Update?

[image]
[image]

ricki-z, 2019-11-14:
Hallo,
kannst du testweise auf die Beta-Version wechseln (in der Konfiguration) und schauen, ob diese stabiler läuft? Wir haben leider noch keinen logischen Grund für das Verhalten einiger SDS011 gefunden, die seit der letzten stabilen Version nicht mehr durchgängig Daten liefern.

iceweasel1, 2019-11-14:
Hallo @ricki-z,

hatte heute früh, circa eine dreiviertel Stunde nach deinem Kommentar es so umgesetzt. Habe bei der Konfiguration den Hacken bei "Auto-Update" herausgenommen und darunter bei "Lade Beta Versionen" eingesetzt. Danach einen Neustart. Die erste halbe Stunde hatte ich es beobachtet. Es blieb weiterhin bei den Bindestrichen. Danach nochmals und noch ein weiteres Mal über die Übersichtsfläche neugestartet. Nichts. Zwischenzeitlich hat sich der Sensor anscheinend auch mehrfach von selbst neugestartet. (Da jetzt gerade 14 Messungen steht und bis vor Stunden noch 250 ca.)

Jetzt habe ich wieder reingeschaut und sehe, das es anscheinend seit kurzem mal etwas zu funktionieren scheint. Zumindest seh ich im Archiv eine Messung die seit kurzer Zeit stattfindet. Die letzten 14 Stunden circa, blieben aber leider leer.

Eben während des Beitrags, Messung 14 Bindestriche, Messung 15 mit Zahlen. Danach die nächsten 4 - 5 wieder nur Bindestriche.

https://www.madavi.de/sensor/graph.php?sensor=esp8266-331968-sds011

Grüße
[image]
[image]