opendata-stuttgart/meta #124: Lärmsensor via API abfragen ist nicht dokumentiert

DasNordlicht, 2019-12-02:
Um die Lärm Sensoren in Deutschlan abzufragen kann man ja:

https://data.sensor.community/airrohr/v1/filter/country=DE&type=Laerm

anfragen. Nur es ist leider nicht beschrieben.

ricki-z, 2019-12-02:
Lärm ist noch seeeehr Beta, weshalb wir im Moment Abfragen noch nicht offiziell unterstützen möchten. Aktuell gibt es auch nur 3 oder 4 Lärmsensoren, die alle Testsysteme sind.
Sobald wir aus der Beta-Phase raus sind, wird die Beschreibung ergänzt. Und ich sehe gerade, daß wir zumindest den "type" noch ändern werden. Lärm ist ja eher das Phänomen denn der Typ des Sensors.

quantenquark, 2024-11-08:
Moin. Ich habe seit Jahren mehrere Sensoren in Betrieb und füttere die Sensor.Community mit BME/SDS und DNMS-Daten.
Alle Sensoren werden jetzt zentral in meine Wetter/Messstation im Garten integriert und ich nutze HomeAssistant mit dem Add-On Sensor.Community.
Leider wird die Sensor-ID des DNMS weiterhin nicht unterstützt was ich sehr schade finde.

Ist geplant das Add-on entsprechend zu erweitern? Kann ich damit noch rechnen?

ricki-z, 2024-12-30:
@quantenquark Hallo, das HomeAssistant Add-On wird nicht von uns betreut. Daher können wir dazu nichts sagen.